tensorflow/tflite-micro #1226: Micro Speech Example Deploy to STM32F746 error!

FragrantRookie, 2022-06-27:
Following the method to build this example,

https://github.com/tensorflow/tflite-micro/issues/475#issuecomment-915649543

error reported like the following:
lmk@iZrj92rskh5qft0udfmrvnZ:~/code/tflite-micro/test/mbed-os$ mbed compile -m DISCO_F746NG -t GCC_ARM -D CMSIS_NN
[mbed] Working path "/home/lmk/code/tflite-micro/test/mbed-os" (library)
[mbed] Program path "/home/lmk/code/tflite-micro/test"
[mbed] WARNING: Missing Python modules were not auto-installed.
       The Mbed OS tools in this program require the following Python modules: future, click
       You can install all missing modules by running "pip install -r requirements.txt" in "/home/lmk/code/tflite-micro/test/mbed-os"
       On Posix systems (Linux, etc) you might have to switch to superuser account or use "sudo"
---
Building project test (DISCO_F746NG, GCC_ARM)
Scan: test
Compile [  0.2%]: micro_model_settings.cc
Compile [  0.3%]: command_responder.cc
Compile [  0.3%]: main.cc
Compile [  0.4%]: audio_provider.cc
Compile [  0.5%]: no_micro_features_data.cc
Compile [  0.6%]: feature_provider.cc
Compile [  0.7%]: yes_micro_features_data.cc
Compile [  0.8%]: no_30ms_audio_data.cc
Compile [  0.8%]: yes_30ms_audio_data.cc
Compile [  0.9%]: micro_speech_model_data.cc
Compile [  1.0%]: recognize_commands.cc
Compile [  1.1%]: RTX_Config.c
Compile [  1.2%]: no_1000ms_audio_data.cc
Compile [  1.3%]: cmsis_os1.c
[Fatal Error] base.h@38,10: cstdint: No such file or directory
[ERROR] In file included from ../third_party/flatbuffers/include/flatbuffers/string.h:20,
                 from ../mbed-os/cmsis/CMSIS_5/CMSIS/RTOS2/RTX/Library/cmsis_os1.c:27:
../third_party/flatbuffers/include/flatbuffers/base.h:38:10: fatal error: cstdint: No such file or directory
   38 | #include <cstdint>
      |          ^~~~~~~~~
compilation terminated.

[mbed] ERROR: "/usr/bin/python3" returned error.
       Code: 1
       Path: "/home/lmk/code/tflite-micro/test/mbed-os"
       Command: "/usr/bin/python3 -u /home/lmk/code/tflite-micro/test/mbed-os/tools/make.py -D CMSIS_NN -t GCC_ARM -m DISCO_F746NG --source .. --build ../BUILD/DISCO_F746NG/GCC_ARM"
       Tip: You could retry the last command with "-v" flag for verbose output

DonaldWeier, 2022-08-18:
Were you able to resolve this? I am running into this exact issue (missing cstdint). I'm hoping to get the example working and update the README with the appropriate steps once I understand all the setup required.

FragrantRookie, 2022-08-19:
> Were you able to resolve this? I am running into this exact issue (missing cstdint). I'm hoping to get the example working and update the README with the appropriate steps once I understand all the setup required.

I didn't solve it. Please ask questions in the tensorflow mainline, not in the micro branch.
https://github.com/tensorflow/tensorflow/issues

Terryvanderjagt, 2023-03-01:
This issue is happening because mbed tries to compile it's C files with the string.h file of flatbuffers instead of the libc string header. It then fails to include cstdint because this is a C++ header. I resolved this issue by appending fb_ to all flatbuffer header files except for flatbuffers.h and flexbuffers.h with the shell script below. I hope this will help someone running into the same issue.

```
for f in *; do
	if [[ "$f" == "flatbuffers.h" || "$f" == "flexbuffers.h" ]]; then
		continue
	fi

	mv "$f" "fb_$f"
	sed -i "s@#include \"flatbuffers/$f@#include \"flatbuffers/fb_$f@g" *
done
```

closed 2023-05-05 by github-actions[bot]