tensorflow/tflite-micro #1243: LSTM Support for the TFLM

1msec, 2022-07-06:
Is there any plan to support lstm for the TFLM?

giovanni6inve, 2022-07-14:
Add   AddUnidirectionalSequenceLSTM(); in  all_ops_resolver.cpp file and after it works:)

1msec, 2022-07-18:
Thank you for your reply!
Let me ask more about it.

I made my own model using LSTM cell.
And I converted it into a tflite model using the convert guide.
I got the below after converting.
I found the there were no LSTM(or RNN) layers in the model.

Here are my questions.
1. Which layer or operation is matched to the "UnidirectionalSequenceLSTM" you mentioned.
   (Let me know so I can change my Keras Model, I'd like to use LSTM)
2. Could you teach me which layer or operation is matched to LSTM Cell that I used in my Keras model.

> **My Model** 
    my_model = tf.keras.Sequential()
    cell = tf.keras.layers.LSTMCell(units=num_classes, input_shape=(sequence_length, input_dim))
    my_model.add(tf.keras.layers.RNN(cell=cell, return_sequences=True))
    my_model.add(tf.keras.layers.TimeDistributed(tf.keras.layers.Dense(units=num_classes, activation='softmax')))
    my_model.compile(loss='categorical_crossentropy', 
                    optimizer=tf.keras.optimizers.Adam(learning_rate=learning_rate),
                    metrics=['accuracy'])

> **tflite converted** 
    [image]

Thank you in advance :)

christophsiegl, 2022-08-30:
> Add AddUnidirectionalSequenceLSTM(); in all_ops_resolver.cpp file and after it works:)

@giovanni6inve  I tried out this and it works (at least on linux_x86_64), but it takes quiet a long time to run! But why is this opcode not added per default in all_ops_resolver.cc? This makes no sense to me? 

>     1. Which layer or operation is matched to the "UnidirectionalSequenceLSTM" you mentioned.
>        (Let me know so I can change my Keras Model, I'd like to use LSTM)

@1msec  Try to use keras.layers.LSTM()! This should work.

PR #1388 (2022-09-08): Add MNIST LSTM model evaulation
# Evaluate trained LSTM model on MNIST dataset 

## evaluate.py

Given the trained .tflite model and a MNIST image path, output the predicted category (digit 0-9) probability. 

Example:
`bazel build tensorflow/lite/micro/examples/mnist_lstm:evaluate`
`bazel-bin/tensorflow/lite/micro/examples/mnist_lstm/evaluate --model_path='.tflite file path' --img_path='MNIST image path'`

## evaluate_test.py

Test the evaluation code under the following criteria: 
* Using random input to compare results from TFLite interpreter's results
* Using sample MNIST images to test mo [... cut]
commit 945d1976 (2022-09-08): Add MNIST LSTM model evaulation

# Evaluate trained LSTM model on MNIST dataset 

## evaluate.py

Given the trained .tflite model and a MNIST image path, output the predicted category (digit 0-9) probability. 

Example:
`bazel build tensorflow/lite/micro/examples/mnist_lstm:evaluate`
`bazel-bin/tensorflow/lite/micro/examples/mnist_lstm/evaluate --model_path='.tflite file path' --img_path='MNIST image path'`

## evaluate_test.py

Test the evaluation code under the following criteria: 
* Using random input to compare results from TFLite interpreter's results
* Using sample MNIST images to test m [... cut]
closed 2022-09-08 by mergify[bot]