tensorflow/tflite-micro #1249: Failed to allocate temp memory.

amkmpro, 2022-07-09:
Hello, I want to run a MobileNetV1 on tflite-micro and use the person-detection example. But I have an error and it's Failed to allocate temp memory. I have increased tensor_arena_size (in person_detection_test.cc under person_detection directory) but this problem doesn't solve. The strange thing is that when I increase tensor_arena_size, the memory available decreases. In below, I bring some results.
```
tenor_arena_size = 140*1024; //available = 456
tenor_arena_size = 141*1024; //available = 1480
tenor_arena_size = 142*1024; //available = 2504
tenor_arena_size = 143*1024; //available = 3528
tenor_arena_size = 144*1024; //available = 616
```

Could you explain it?

Thanks in advance.

rockyrhodes, 2022-07-18:
When you report available memory above, where do those numbers come from? Is this from the error message from a failure to allocate a temp buffer? If so, what is probably happening is that with an arena size of 144*1024, you now have a big enough arena to allocate the temporary buffer that failed with the previous 4 sizes, but a later temporary allocation still fails. You need to increase the size further until all temporary buffer allocation requests can be satisfied from the arena.

amkmpro, 2022-07-21:
Thank you for your response. These numbers are from the error message. When I set tensor_arena_size=144*1024, available memory is reduced while compared with 143*1024, memory available is reduced. 
(Note: I run the example with one of the tensor_arena_size separately.)

closed 2023-05-04 by github-actions[bot]