tensorflow/tflite-micro #1271: Unable to use TFLM with ArduinoBLE on Xiao BLE Sense/Arduino Nano 33 BLE Sense

HieronymusLex, 2022-07-21:
Following on from https://github.com/tensorflow/tensorflow/issues/41271 I've observed the same behaviour described in that issue i.e. attempting to use TFLM on an Arduino Nano 33 BLE Sense (and now a Xiao BLE Sense) when the model is defined `const unsigned char model[]` whilst **also** attempting to use the `ArduinoBLE` module causes the system to "hard fault". By hard fault I mean it just hangs, execution stops, no serial output is printed and the board needs to be reset. Removing the `const` appears to fix the issue so my understanding from the referenced issue is that it's something related to storing the model in flash? I've tried various models which all work fine on their own when not using ArduinoBLE

Note without using `ArduinoBLE`, the same model works fine and the `ArduinoBLE` lib works fine in the absence of TFLM. The issue is only observed when using them together.

I'd like to debug it further but I'm not sure where to start. A possible lead:

> One offhand thought is that the BLE library may implicitly reserve certain addresses for BLE registers and DMA, and these could potentially conflict with the addresses used to access flash. In this case, updating the memory map could help.

rkuester, 2022-08-19:
Assigning to @ddavis-2015 for comment, since David's worked on something related to this.

ddavis-2015, 2022-09-01:
I have just tested the magic_wand example on the Nano 33 BLE Sense, with no issues.  Both the inference operations and Bluetooth are working correctly.  ArduinoBLE library version 1.2.1 was used.  If there are issues with more recent versions of the ArduinoBLE library, the TFLM team will be unable to resolve such issues, as that library is not part of TFLM.

The model (i.e. flatbuffer) can and should be stored in flash memory.  If an operator in your model is incorrectly manipulating the flatbuffer, we would like to know about it.  Attaching your model (as a converted .tflite file) would allow us to inspect which operators are being used.  If you decide to share your model with us, please feel free to reopen this issue.

Please be aware that [tflite-micro-arduino-examples](https://github.com/tensorflow/tflite-micro-arduino-examples) is now the official repo. for Arduino examples of TFLM usage.  Any problems with TFLM on the Nano 33 BLE Sense must originate from that code base.

closed 2022-09-01 by ddavis-2015

HieronymusLex, 2022-09-07:
Thanks for taking a look @ddavis-2015 ! I ran the latest version of magic wand notebook and can see that when the model is converted to a C source file, the resulting model is now defined as `unsigned char g_magic_wand_model_data[] = ...` which as you say works as expected and is inline with my findings I put in the issue description. However, if you modify `magic_wand_model_data.cc` to change L1 to `const unsigned char g_magic_wand_model_data[] = ...` then the issue should be reproducible when you invoke the model.

For my use case this is fine, but it was just unexpected behaviour so seemed worth flagging. It may be that ArduinoBLE is the "culprit" which of course is beyond the control of the TFLM team