tensorflow/tflite-micro #1283: Cannot run tflm python interpreter test with bazel

rcepeda95, 2022-07-25:
I'm gettting some numpy errors when I run the bazel test tensorflow/lite/micro/tools/python_interpreter/tests:interpreter_test.

Steps:
```
1. git clone tfl-micro
2.  bazel test tensorflow/lite/micro/tools/python_interpreter/tests:interpreter_test
```

OUTPUT:

```
In file included from tensorflow/lite/micro/python/interpreter/src/numpy_utils.cc:16:
./tensorflow/lite/micro/python/interpreter/src/numpy_utils.h:20:10: fatal error: numpy/ndarraytypes.h: No such file or directory
   20 | #include <numpy/ndarraytypes.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~
```

I'm using ...
```
Python 3.10.5
Numpy 1.23.1

```
I realize that I can get the numpy include directory from numpy.get_include() but I have never used bazel and cannot seem to find the right place to add that piece. I also presume that something else needs to be done than just pointing gcc to that include directory.

Any help would be appreciated.

I am trying to get this test running to ultimately debug some different outputs I'm seeing when I run a TFLite model in Python vs that model converted to TFLM on a Raspberry Pi Pico with the same input. I was hoping that I could see if I used this to test the tflmicro model to see if something is wrong with the implementation that's ported over for the Pico.

advaitjain, 2022-07-26:
Can you try with Python 3.9 in a virtual environment.

Start by removing your bazel cache
```
sudo rm -rf ~/.cache/bazel/
```

Set up the virtual environment as described [here](https://github.com/tensorflow/tflite-micro/blob/main/docs/python.md#manual-setup-illustration):
```
# The cloned tflite-micro folder needs to be renamed to tflite_micro
mv tflite-micro tflite_micro
python3.9 -m venv tflite_micro/venv
echo "export PYTHONPATH=\${PYTHONPATH}:${PWD}" >> tflite_micro/venv/bin/activate
cd tflite_micro
source venv/bin/activate
pip install --upgrade pip
pip install -r third_party/requirements.txt
```

Build with bazel from within the virtual environment:
```
source venv/bin/activate
bazel test tensorflow/lite/micro/tools/python_interpreter/tests:interpreter_test
```

rcepeda95, 2022-07-27:
Hey @advaitjain, 

I tried using `pyenv` to change to Python 3.9, but that messed up bazel to find the python include directory. So I installed python3.9 directly and used that to build the venv but I got the very same issue with the `numpy/ndarraytypes.h`

mansnils, 2022-08-15:
I got the same problem when running https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/tools/ci_build/test_bazel.sh, .i.e.  bazel build tensorflow/lite/micro/... \
  --build_tag_filters=-no_oss

closed 2022-10-03 by mergify[bot]