tensorflow/tflite-micro #1297: Training dataset method for Micro speech

LBoutigny, 2022-08-02:
Hello and thank you for reading this comment

I'm quite new to ML in general, I've evaluated and did some tinkering of micro_speech on arduino BLE 33. 

At the moment, I've tried detecting more than 2 words on the plateform (up, left, right, down). I achieved well performance metrics on training using the provided tf scripts (around 90%) so, I was surprised to see of inacurracy of the inference on arduino target. (as reference, you can check out this video https://www.youtube.com/watch?v=ImrYfkDbC_0 which is more or less what I've experienced ) 

I had this tought, what if I used recorded samples from the embedded microphone for the dataset training instead of the pretty ones we use on our laptops ?

Shouldn't it be more representative of what the ML system should learn ?

Cheers 

LB

advaitjain, 2022-08-08:
Assigning to @ddavis-2015 to answer Arduino related questions.

ddavis-2015, 2022-08-31:
Was the version of TensorFlow Lite Micro you are using from the Arduino-Examples repository?  Or from the Arduino IDE library download?  The Arduino-Examples repository had the micro-speech example updated in Dec. 2021 to fix some issues with the code base that affected recognition accuracy.

You are more than welcome and encouraged to bring your own data for training your model.

closed 2022-08-31 by ddavis-2015