tensorflow/tflite-micro #1322: Reshape error (num_input_elements != num_output_elements) when using LSTM layer on ESP32

ndaneil, 2022-08-06:
I exported a TFLite model containing a 40-cell LSTM layer:
`lstm_100 (LSTM)              (None, 40)                8160`
`dense_1300 (Dense)           (None, 11)                451`
 In TFLite Micro on the ESP32, I get the following errors during `AllocateTensors();`:
`lib/tflite-lib/tensorflow/lite/micro/kernels/reshape.cc:76 num_input_elements != num_output_elements (560 != 40)
lib/tflite-lib/tensorflow/lite/micro/kernels/reshape.cc:76 num_input_elements != num_output_elements (560 != 40)
lib/tflite-lib/tensorflow/lite/micro/kernels/reshape.cc:86 ReshapeOutput(context, node) != kTfLiteOk (1 != 0)
Node RESHAPE (number 7f) failed to prepare with status 1`
I am using [this](https://github.com/espressif/tflite-micro-esp-examples) tflite-micro version (master branch)

closed 2023-04-27 by github-actions[bot]

YuriSizuku, 2024-01-24:
Have you solved this problem ?

ndaneil, 2024-01-26:
Unfortunately I could not get it to work.