tensorflow/tflite-micro #1337: tensorflow/lite/micro/tools/ci_build/test_x86.sh does not compile with gcc 12.1

mansnils, 2022-08-11:
gcc-12.1 warns that size of metadata->name()->c_str() (it sees as 0) is smaller than strlen(kOfflineMemAllocMetadata) (which is 23):
https://github.com/tensorflow/tflite-micro/blob/eacc122d917605288e694220a5074c17750292f7/tensorflow/lite/micro/micro_allocation_info.cc#L326

Error:
tensorflow/lite/micro/micro_allocation_info.cc: In member function ‘TfLiteStatus tflite::AllocationInfoBuilder::GetOfflinePlannedOffsets(const int32_t**)’:
tensorflow/lite/micro/micro_allocation_info.cc:326:18: error: ‘int strncmp(const char*, const char*, size_t)’ specified bound 23 exceeds source size 0 [-Werror=stringop-overread]

advaitjain, 2022-11-10:
With https://github.com/tensorflow/tflite-micro/pull/1477 we are seeing an error with gcc-12 again.

https://github.com/tensorflow/tflite-micro/pull/1544 is suppressing the warning for now, but it would be good to see what better fix would be.

We'll also update CI to use gcc-12.

advaitjain, 2022-11-18:
Closing this issue since it should be fixed with #1550. Let's reopen if it is still not fixed.

closed 2022-11-18 by advaitjain