tensorflow/tflite-micro #1370: configuring weight layout alignment in memory

Jconn, 2022-08-22:
Hello,

I'm running tensorflow lite micro on a platform with 16 byte aligned memory accesses. While I was inspecting performance of the lstm layer, I see that some of my weights matrices are not aligned to 16 bytes - which incurs a performance hit. Is there any method in the tflite micro model conversion process that lets me align weights to memory boundaries?

petewarden, 2022-08-27:
Just as a note, this script is doing some alignment for weights, though I don't remember all the details: https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/tools/tflite_flatbuffer_align.cc

Jconn, 2022-09-12:
that script does align weights to 16 bytes! When I check the [schema](https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/schema/schema.fbs#L1260), I see the buffer definition force_align attribute is 16. I think this is what gives me my 16 byte alignment.

closed 2022-09-12 by Jconn