tensorflow/tflite-micro #1371: mac os monterey: micro speech example doesn't work.

Raunak-Singh-Inventor, 2022-08-22:
I ran the following commands in terminal:
```
make -f tensorflow/lite/micro/tools/make/Makefile micro_speech
tensorflow/lite/micro/tools/make/gen/osx_x86_64_default/bin/micro_speech
```
and nothing happens. The micro speech example doesn't run and it doesn't ask me for any microphone access.

Please help in any way.

Output I get from running the first command:
```
tensorflow/lite/micro/tools/make/downloads/flatbuffers already exists, skipping the download.
tensorflow/lite/micro/tools/make/downloads/kissfft already exists, skipping the download.
tensorflow/lite/micro/tools/make/downloads/pigweed already exists, skipping the download.
find: tensorflow/lite/micro/tools/gen_micro_mutable_op_resolver_test/person_detect: No such file or directory
g++ -std=c++11 -fno-rtti -fno-exceptions -fno-threadsafe-statics -Werror -fno-unwind-tables -ffunction-sections -fdata-sections -fmessage-length=0 -DTF_LITE_STATIC_MEMORY -DTF_LITE_DISABLE_X86_NEON -Wsign-compare -Wdouble-promotion -Wshadow -Wunused-variable -Wunused-function -Wswitch -Wvla -Wall -Wextra -Wmissing-field-initializers -Wstrict-aliasing -Wno-unused-parameter  -DTF_LITE_USE_CTIME -Os -I. -Itensorflow/lite/micro/tools/make/downloads/gemmlowp -Itensorflow/lite/micro/tools/make/downloads/flatbuffers/include -Itensorflow/lite/micro/tools/make/downloads/ruy -Itensorflow/lite/micro/tools/make/gen/osx_x86_64_default/genfiles/ -Itensorflow/lite/micro/tools/make/downloads/kissfft -c tensorflow/lite/micro/tools/make/gen/osx_x86_64_default/genfiles/tensorflow/lite/micro/examples/micro_speech/micro_speech_model_data.cc -o tensorflow/lite/micro/tools/make/gen/osx_x86_64_default/obj/core/tensorflow/lite/micro/tools/make/gen/osx_x86_64_default/genfiles/tensorflow/lite/micro/examples/micro_speech/micro_speech_model_data.o
g++ -std=c++11 -fno-rtti -fno-exceptions -fno-threadsafe-statics -Werror -fno-unwind-tables -ffunction-sections -fdata-sections -fmessage-length=0 -DTF_LITE_STATIC_MEMORY -DTF_LITE_DISABLE_X86_NEON -Wsign-compare -Wdouble-promotion -Wshadow -Wunused-variable -Wunused-function -Wswitch -Wvla -Wall -Wextra -Wmissing-field-initializers -Wstrict-aliasing -Wno-unused-parameter  -DTF_LITE_USE_CTIME -Os -I. -Itensorflow/lite/micro/tools/make/downloads/gemmlowp -Itensorflow/lite/micro/tools/make/downloads/flatbuffers/include -Itensorflow/lite/micro/tools/make/downloads/ruy -Itensorflow/lite/micro/tools/make/gen/osx_x86_64_default/genfiles/ -Itensorflow/lite/micro/tools/make/downloads/kissfft -c tensorflow/lite/micro/tools/make/gen/osx_x86_64_default/genfiles/tensorflow/lite/micro/examples/micro_speech/testdata/no_1000ms_audio_data.cc -o tensorflow/lite/micro/tools/make/gen/osx_x86_64_default/obj/core/tensorflow/lite/micro/tools/make/gen/osx_x86_64_default/genfiles/tensorflow/lite/micro/examples/micro_speech/testdata/no_1000ms_audio_data.o
g++ -std=c++11 -fno-rtti -fno-exceptions -fno-threadsafe-statics -Werror -fno-unwind-tables -ffunction-sections -fdata-sections -fmessage-length=0 -DTF_LITE_STATIC_MEMORY -DTF_LITE_DISABLE_X86_NEON -Wsign-compare -Wdouble-promotion -Wshadow -Wunused-variable -Wunused-function -Wswitch -Wvla -Wall -Wextra -Wmissing-field-initializers -Wstrict-aliasing -Wno-unused-parameter  -DTF_LITE_USE_CTIME -Os -I. -Itensorflow/lite/micro/tools/make/downloads/gemmlowp -Itensorflow/lite/micro/tools/make/downloads/flatbuffers/include -Itensorflow/lite/micro/tools/make/downloads/ruy -Itensorflow/lite/micro/tools/make/gen/osx_x86_64_default/genfiles/ -Itensorflow/lite/micro/tools/make/downloads/kissfft -c tensorflow/lite/micro/tools/make/gen/osx_x86_64_default/genfiles/tensorflow/lite/micro/examples/micro_speech/testdata/no_30ms_audio_data.cc -o tensorflow/lite/micro/tools/make/gen/osx_x86_64_default/obj/core/tensorflow/lite/micro/tools/make/gen/osx_x86_64_default/genfiles/tensorflow/lite/micro/examples/micro_speech/testdata/no_30ms_audio_data.o
g++ -std=c++11 -fno-rtti -fno-exceptions -fno-threadsafe-statics -Werror -fno-unwind-tables -ffunction-sections -fdata-sections -fmessage-length=0 -DTF_LITE_STATIC_MEMORY -DTF_LITE_DISABLE_X86_NEON -Wsign-compare -Wdouble-promotion -Wshadow -Wunused-variable -Wunused-function -Wswitch -Wvla -Wall -Wextra -Wmissing-field-initializers -Wstrict-aliasing -Wno-unused-parameter  -DTF_LITE_USE_CTIME -Os -I. -Itensorflow/lite/micro/tools/make/downloads/gemmlowp -Itensorflow/lite/micro/tools/make/downloads/flatbuffers/include -Itensorflow/lite/micro/tools/make/downloads/ruy -Itensorflow/lite/micro/tools/make/gen/osx_x86_64_default/genfiles/ -Itensorflow/lite/micro/tools/make/downloads/kissfft -c tensorflow/lite/micro/tools/make/gen/osx_x86_64_default/genfiles/tensorflow/lite/micro/examples/micro_speech/testdata/yes_1000ms_audio_data.cc -o tensorflow/lite/micro/tools/make/gen/osx_x86_64_default/obj/core/tensorflow/lite/micro/tools/make/gen/osx_x86_64_default/genfiles/tensorflow/lite/micro/examples/micro_speech/testdata/yes_1000ms_audio_data.o
g++ -std=c++11 -fno-rtti -fno-exceptions -fno-threadsafe-statics -Werror -fno-unwind-tables -ffunction-sections -fdata-sections -fmessage-length=0 -DTF_LITE_STATIC_MEMORY -DTF_LITE_DISABLE_X86_NEON -Wsign-compare -Wdouble-promotion -Wshadow -Wunused-variable -Wunused-function -Wswitch -Wvla -Wall -Wextra -Wmissing-field-initializers -Wstrict-aliasing -Wno-unused-parameter  -DTF_LITE_USE_CTIME -Os -I. -Itensorflow/lite/micro/tools/make/downloads/gemmlowp -Itensorflow/lite/micro/tools/make/downloads/flatbuffers/include -Itensorflow/lite/micro/tools/make/downloads/ruy -Itensorflow/lite/micro/tools/make/gen/osx_x86_64_default/genfiles/ -Itensorflow/lite/micro/tools/make/downloads/kissfft -c tensorflow/lite/micro/tools/make/gen/osx_x86_64_default/genfiles/tensorflow/lite/micro/examples/micro_speech/testdata/yes_30ms_audio_data.cc -o tensorflow/lite/micro/tools/make/gen/osx_x86_64_default/obj/core/tensorflow/lite/micro/tools/make/gen/osx_x86_64_default/genfiles/tensorflow/lite/micro/examples/micro_speech/testdata/yes_30ms_audio_data.o
g++ -std=c++11 -fno-rtti -fno-exceptions -fno-threadsafe-statics -Werror -fno-unwind-tables -ffunction-sections -fdata-sections -fmessage-length=0 -DTF_LITE_STATIC_MEMORY -DTF_LITE_DISABLE_X86_NEON -Wsign-compare -Wdouble-promotion -Wshadow -Wunused-variable -Wunused-function -Wswitch -Wvla -Wall -Wextra -Wmissing-field-initializers -Wstrict-aliasing -Wno-unused-parameter  -DTF_LITE_USE_CTIME -I. -Itensorflow/lite/micro/tools/make/downloads/gemmlowp -Itensorflow/lite/micro/tools/make/downloads/flatbuffers/include -Itensorflow/lite/micro/tools/make/downloads/ruy -Itensorflow/lite/micro/tools/make/gen/osx_x86_64_default/genfiles/ -Itensorflow/lite/micro/tools/make/downloads/kissfft -o tensorflow/lite/micro/tools/make/gen/osx_x86_64_default/bin/micro_speech tensorflow/lite/micro/tools/make/gen/osx_x86_64_default/obj/core/tensorflow/lite/micro/examples/micro_speech/main.o tensorflow/lite/micro/tools/make/gen/osx_x86_64_default/obj/core/tensorflow/lite/micro/examples/micro_speech/main_functions.o tensorflow/lite/micro/tools/make/gen/osx_x86_64_default/obj/core/tensorflow/lite/micro/examples/micro_speech/audio_provider.o tensorflow/lite/micro/tools/make/gen/osx_x86_64_default/obj/core/tensorflow/lite/micro/examples/micro_speech/feature_provider.o tensorflow/lite/micro/tools/make/gen/osx_x86_64_default/obj/core/tensorflow/lite/micro/examples/micro_speech/micro_features/no_micro_features_data.o tensorflow/lite/micro/tools/make/gen/osx_x86_64_default/obj/core/tensorflow/lite/micro/examples/micro_speech/micro_features/yes_micro_features_data.o tensorflow/lite/micro/tools/make/gen/osx_x86_64_default/obj/core/tensorflow/lite/micro/examples/micro_speech/recognize_commands.o tensorflow/lite/micro/tools/make/gen/osx_x86_64_default/obj/core/tensorflow/lite/micro/examples/micro_speech/command_responder.o tensorflow/lite/micro/tools/make/gen/osx_x86_64_default/obj/core/tensorflow/lite/micro/examples/micro_speech/micro_features/micro_features_generator.o tensorflow/lite/micro/tools/make/gen/osx_x86_64_default/obj/core/tensorflow/lite/micro/examples/micro_speech/micro_features/micro_model_settings.o tensorflow/lite/micro/tools/make/gen/osx_x86_64_default/obj/core/tensorflow/lite/experimental/microfrontend/lib/fft.o tensorflow/lite/micro/tools/make/gen/osx_x86_64_default/obj/core/tensorflow/lite/experimental/microfrontend/lib/fft_util.o tensorflow/lite/micro/tools/make/gen/osx_x86_64_default/obj/core/tensorflow/lite/experimental/microfrontend/lib/filterbank.o tensorflow/lite/micro/tools/make/gen/osx_x86_6 [... cut ...] flow/lite/micro/tools/make/gen/osx_x86_64_default/obj/core/tensorflow/lite/micro/tools/make/gen/osx_x86_64_default/genfiles/tensorflow/lite/micro/examples/micro_speech/testdata/yes_1000ms_audio_data.o tensorflow/lite/micro/tools/make/gen/osx_x86_64_default/obj/core/tensorflow/lite/micro/tools/make/gen/osx_x86_64_default/genfiles/tensorflow/lite/micro/examples/micro_speech/testdata/yes_30ms_audio_data.o tensorflow/lite/micro/tools/make/gen/osx_x86_64_default/lib/libtensorflow-microlite.a  -lm
```

Raunak-Singh-Inventor, 2022-08-24:
TFLM Team, I am still waiting for any response. My project is on hold because of this issue.

mandeepsingh-private, 2022-08-24:
Hey folks I am also facing the same issue. Can you please help? How to get past this issue?

eric0708, 2022-12-30:
Hello, I also came across this issue when working through the examples. I found out that the default `audio_provider.cc` originally provided by TensorFlow has default behavior of setting input features of all zero. Also, slight modifications have to be made to Makefile.inc in order for the application to run on MacOS. I have provide the modified `audio_provider.cc` and `Makefile.inc` files in the directory below, simply replace the corresponding files and follow the original steps. Everything should work fine!
[https://github.com/eric0708/TinyML_Projects/tree/main/Micro_Speech](https://github.com/eric0708/TinyML_Projects/tree/main/Micro_Speech)

closed 2023-04-27 by github-actions[bot]