tensorflow/tflite-micro #1372: Is this library still active?

Raunak-Singh-Inventor, 2022-08-22:
I have noticed that issues from more than 1 month ago have not been resolved.

Is this repo still active? Or has TensorFlow shelfed this project?

Thanks in advance for your reply.

advaitjain, 2022-08-22:
Yes, the project is very much active with changes being made by engineers at Google and community contributors. Please see the [commit history](https://github.com/tensorflow/tflite-micro/commits/main) to get a sense of the current activity on the project.

closed 2022-08-22 by advaitjain

Raunak-Singh-Inventor, 2022-08-22:
Thanks for the reply. That's great to hear.

I just wanted to know why many of the issues in this repo aren't being answered.

I am looking to use TFLite Micro in my project, and am looking for a solution to my issue with Micro Speech example: https://github.com/tensorflow/tflite-micro/issues/1371

Raunak-Singh-Inventor, 2022-08-24:
Hi @advaitjain,

Did you find any solution to the issue I mentioned?