tensorflow/tflite-micro #1390: Failed to get registration from op code RANDOM_STANDARD_NORMAL

christophsiegl, 2022-08-31:
Hi,
 
I modified the official micro speech example of tflite-micro to run inference with a custom (VAE) Variational Auto Encoder model on linux_x86_64_default. After replacing the tflite file and modifying the [micro_speech_test.cc](https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/examples/micro_speech/README.md), the make build returns the following error message:
 
```
Didn't find op for builtin opcode 'RANDOM_STANDARD_NORMAL' version '1'. An older version of this builtin might be supported. Are you using an old TFLite binary with a newer model?
 
Failed to get registration from op code RANDOM_STANDARD_NORMAL
```
 
So it seems that the „RANDOM_STANDARD_NORMAL“ is not implemented – at least it does not appear in [all_ops_resolver.cc](https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/all_ops_resolver.cc).

I am using TensorFlow version 2.9.1 and the latest version of tflite-micro framework from 15.8.2022. This operation (= generating random numbers) would be important for several other applications too! Is there something related work in progress or could anyone provide me with an alternative?

Best regards,
Christoph

deepaksharma36, 2022-11-03:
@advaitjain How we can add custom ops like RANDOM_STANDARD_NORMAL to the all_ops_resolver.cc?
@christophsiegl did you find any solution for this issue?

christophsiegl, 2022-11-16:
@deepaksharma36 I did not find any solution so far. But there exists a porting reference which describes how to port TF Lite opcode to TF Lite Micro: https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/docs/porting_reference_ops.md
So far I have not tried to do this by myself ...

closed 2023-04-27 by github-actions[bot]