tensorflow/tflite-micro #1422: Tflite Micro Hello-world continuously restart on ESP32-S3

priyanshu-makhiyaviya, 2022-09-16:
@tensorflow/micro

System information

Host OS Platform and Distribution (e.g., Linux Ubuntu): Ubuntu 18.04
TensorFlow installed from (source or binary): source
Tensorflow version (commit SHA if source): 2.3.0
Target platform (e.g. Arm Mbed OS, Arduino Nano 33 etc.): ESP32-S3

Describe the problem

I am trying to setup the ESP32-S3 for TFlite.
I have compiled TFLite hello-world for ESP32-S3. The compilation got successful but when I flash the image on ESP32-S3, it continuously restarted the board.

Please let me know how can i build and flash TFLite hello-world for ESP32-S3.

Please provide the exact sequence of commands/steps when you ran into the problem

Clone the tensorflow repository:
git clone https://github.com/tensorflow/tensorflow.git -b r2.3

Generate the examples The example project can be generated with the following command:
make -f tensorflow/lite/micro/tools/make/Makefile TARGET=esp generate_hello_world_esp_project

Building the example Go the the example project directory:
cd tensorflow/lite/micro/tools/make/gen/esp_xtensa-esp32/prj/hello_world/esp-idf

Set the target:
idf.py set-target esp32s3

Then build with idf.py:
idf.py build

Flashing an image:
idf.py --port /dev/ttyACM0 flash monitor

But the image when flashed does not work. It keeps restarting.

Any suggestions for this issue?
[cyclic_reboot_ESP32S3.pdf](https://github.com/tensorflow/tflite-micro/files/9581165/cyclic_reboot_ESP32S3.pdf)

surajkumarpandey, 2022-09-26:
Something similar happened with me. I was burning the esp32 firmware and the device went busy after the firmware was uploaded I could not save the hello world file, due to "device busy" message popping up. Any leads for this?

vikramdattu, 2022-09-27:
Hi @priyanshu-makhiyaviya is this a solved problem? You have a similar issue raised on espressif's `tflite-micro-esp-examples` repo and it seems to resolved for you.

If the problem is still seen, please share the logs for the same.

priyanshu-makhiyaviya, 2022-09-27:
Hi @vikramdattu ,
I was trying with both repo , _tensorfloe/tflite-micro_ and _tflite-micro-esp-examples_
The issue raised on espressif's _tflite-micro-esp-examples_ repo got resolved.
This is the issue with the _tensorfloe/tflite-micro_ repo. The log is already attached. 

But anyway I got a working example with the _tflite-micro-esp-examples_ and currently not focusing on _tensorfloe/tflite-micro_

Thank you.

vikramdattu, 2022-09-27:
Marking the issue as resolved since, it is no more a problem for you.

closed 2022-09-27 by vikramdattu