tensorflow/tflite-micro #1447: CMSIS-NN making tflite inference slower for fully connected layers

JgazteluML, 2022-09-28:
I was making some tests to learn about the behaviour of the NN when using CMSIS-DSP and CMSIS-NN with different projects and I’ve noticed that when I'm not using CMSIS-NN all the inferences in int8 are much slower but when used on fully connected layers the inference time drops. On a NN with 3 dense layers of 512 neurons each time results were 598ms with CMSIS-NN and 516ms without it (I’d like to add that in those cases I just wanted to have a large NN to appreciate the inference time I was not looking for good results even it ended giving good results and also that the CPU being used is a CortexM4).
Is there any reason for that? Why in Conv1D, Conv2D and both MobileNetV1 and MobileNetV2 inferences are accelerated when using CMSIS-NN but slowed for fully connected layers.
Thanks
Josu

mansnils, 2022-09-28:
Hi @JgazteluML
How are you running the inferences? I was not aware of this and would like to reproduce it. If you could share the fully connected model that would also help.

JgazteluML, 2022-09-29:
I'm using the CMSIS pack generated by Edge Impulse and using parts of their example code:
`size_t buf_len = sizeof(signalbuff) / sizeof(signalbuff[0]);
			
		  	// Assign callback function to fill buffer used for preprocessing/inference
		  	Signal.total_length = EI_CLASSIFIER_DSP_INPUT_FRAME_SIZE;
		  	Signal.get_data = &get_signal_data;

		  	// Perform DSP pre-processing and inference
		  	res = run_classifier(&Signal, &result, false);` 

and the get_signal_data code is here:
`static int get_signal_data(size_t offset, size_t length, float *out_ptr) {
	memcpy(out_ptr, signalbuff + offset, length * sizeof(float));
	return 0;
}`
Im also adding the C++ code generated for deployment: 
[fullyconnectedbig-v14.zip](https://github.com/tensorflow/tflite-micro/files/9671800/fullyconnectedbig-v14.zip)

mansnils, 2022-09-29:
Thanks. I noticed the arm_fully_connected_s8.c in the zip file you provided is almost 2 years old, it is from October 2020. Maybe you could try the latest CMSIS-NN? Also could you share the standalone fully connected model (or just layer) in source code format or as a tflite file?

JgazteluML, 2022-09-29:
Where can I get the latest CMSIS-NN version?

JgazteluML, 2022-09-30:
I would love to do so but I really dont know where can I see that, the only thing I know is that when setting the NN architecture I used three fully connected layers of 512 neurons with no dropout so it has a "big" size where I can compare the differences between using CMSIS-NN and not using it.
Here is the screen capture of the network design on Edge Impulse:
[image]
I will try inferencing with the new libraries and I'll tell you the results

closed 2023-04-20 by github-actions[bot]