tensorflow/tflite-micro #1455: LSTM kernel in TFL and TFLM use different tanh and sigmoid

mansnils, 2022-10-03:
The LSTM kernel in TFL and TFLM use different implementations of tanh and sigmoid that are not bit exact.

See here for TFLM:
https://github.com/tensorflow/tflite-micro/blob/5818e46c78fcce795077c119336265dcacc0bf70/tensorflow/lite/micro/kernels/lstm_eval.cc#L537
https://github.com/tensorflow/tflite-micro/blob/5818e46c78fcce795077c119336265dcacc0bf70/tensorflow/lite/micro/kernels/lstm_eval.cc#L626

And TFL:
https://github.com/tensorflow/tensorflow/blob/d415e0d8f95d2250322035e396c35020c45281bc/tensorflow/lite/kernels/lstm_eval.cc#L588
https://github.com/tensorflow/tensorflow/blob/d415e0d8f95d2250322035e396c35020c45281bc/tensorflow/lite/kernels/lstm_eval.cc#L668

UlrikHjort, 2022-10-30:
@vdkhoi @SebastianBoblest FYI

mansnils, 2022-11-23:
Merging this PR to TFL would resolve the issue:  https://github.com/tensorflow/tflite-micro/pull/1253

closed 2023-04-20 by github-actions[bot]