tensorflow/tflite-micro #1492: Fix casting to avoid precision loss when mixing double and floats

UlrikHjort, 2022-10-21:
In some computations floats are used and the result is casted to double. That can result in precision loss  and the correct way is to cast each parameter in the expression to the resulting type. (Should be enough to cast only one parameter but better to be safe and cast all)

advaitjain, 2022-11-08:
@UlrikHjort, @SebastianBoblest, @vdkhoi , @bewagner: I just created two PRs (#1536 and #1537) that revert some of the changes that have already been merged.

The rationale is the following:
@mansnils from the CMSIS-NN team and I just chatted. #1491 and #1501 are actually introducing a difference between Lite and Micro. If we do need these changes they would have to be made in both the TFLM and the TfLite kernels.

Before going down that path (which would involve a reasonable amount of time and effort from everyone involved), we would like to better understand #1492 is and see if that can be reproduced by the CMSIS-NN team as well.

Let's continue the conversation here.

Any information that you can provide to help reproduce the issue that you are seeing would be very helpful.

mansnils, 2022-11-08:
In general I am in favor of all these changes. The concern is that TFLM will divert from TFL. We want TFL and TFLM to be bit-exact. So as @advaitjain mentioned, the changes should then first go into TFL and once merged, then be merged to TFLM.
I think in the best of worlds these changes were merged in both TFL and TFLM. 
Is there some tflite model where TFL and TFLM have different output, and where these changes resolve that? 
If yes, is that so even without the same changes in TFL?

SebastianBoblest, 2022-11-09:
Hi, we totally agree, we should not lose bit-exactness between tflite and tflm. 
The background of this PR originally was: We experienced numerical differences for the pretrained resnet of the ml perf tiny benchmarks for some random inputs. When we started to look for possible reasons, we spotted the tiny bug addressed in #1500. 
There, the same computation is done in two different functions with different casting and we think this clearly is a bug.
It also changes the multiplier in the cmsis backend of tflm a tiny bit.
But to be very clear: Our corrections did not resolve the differences between tflite and tflm, we only thought that our proposed changes are actually more consistent because the final results of computations are always cast to double.
We are very interested in finding the root cause of the deviations tflite to tflm and would gladly help here.

mansnils, 2022-11-10:
What if you apply the same changes in TFL as well? If the difference is related to running TFL and TFLM on different targets, I suspect that will resolve the difference. At least we have seen similar cases like that before. If that is the case we have good reason to merge the changes in both TFL and TFLM.

UlrikHjort, 2022-11-10:
Seems all double castings are correct in TFL except from two which are already covered in our pending PR's:

#1502 in  tensorflow/lite/kernels/internal/reference/reduce.h

and 

#1500 in tensorflow/lite/kernels/kernel_util.cc

mansnils, 2022-11-16:
https://github.com/tensorflow/tflite-micro/pull/1502 and https://github.com/tensorflow/tflite-micro/pull/1500 should then first get merged to the tensorflow repo and then automatically sync to the tflite-micro repo, right?
Assuming that is done, would it improve/fix potential output diffs between TFL and TFLM for some of your use cases related ml perf tiny?

closed 2023-04-18 by github-actions[bot]