tensorflow/tflite-micro #1494: MicroVsnprintf does not interpret "%lu" correctly (micro_profiler.cc) (PRIu32)

tcal-x, 2022-10-22:
In this line in `micro_profiler.cc`:
```
    MicroPrintf("%d,%s,%" PRIu32, i, tags_[i], ticks);
```
`PRIu32` is `lu` on my system (riscv32 toolchain), resulting in a format specification of `%lu`.   However, the output lines look like this:
```
7,DEPTHWISE_CONV_2D,lu
```
that is, the `%lu` is just printed as "`lu`"; the substitution does not take place.

It seems the underlying issue is in MicroVsnprintf().

advaitjain, 2023-04-15:
Sorry for the long delay on this. https://github.com/tensorflow/tflite-micro/pull/1876 has made this change.

closed 2023-04-15 by advaitjain

tcal-x, 2023-04-17:
Hi @advaitjain , ah, I did miss that (and yes, I didn't try retesting before sending my last comment).    Cheers!