tensorflow/tflite-micro #1518: File size increase when converting tflite model

khuangCTS, 2022-11-03:
Hello, 

I wanted to see if there are any ways that I could prevent a file size increase when converting a tflite model to a hex dump. I have 2.5MB tflite model and when converted using "xxd -i "input.tflite" > output.cc" I retrieve a ~15MB file. I was able to work with the xxd command to retrieve a 5MB hex dump by making the output file a continuous stream of hex characters but I would like to know if I can retain the 2.5MB file size. I have an ESP32-S3 with 8MB PSRAM and my current implementation is to add the 5MB hex dump into the SPIFFS filesystem and parsing it within. My understanding is that tflite::getModel requires this hex dump but is there a possible work around with how I compile the tflite model beforehand? A different way to convert my model? I am currently looking into compiling the model for ARM32. Any suggestions or insights would be much appreciated. 

Thanks!

UlrikHjort, 2022-11-03:
Is it the actual .output.cc file that is 15MB ? What should matter is the size of the converted buffer which is located in the end of the output.cc file like e.g unsigned int  <MODEL>_tflite_len = 123456;  

That size should be equal to your tflite model file.

khuangCTS, 2022-11-03:
> Is it the actual .output.cc file that is 15MB ? What should matter is the size of the converted buffer which is located in the end of the output.cc file like e.g unsigned int MODELNAME_tflite_len = 123456;

xxx -i model.tflite > model.cc
[image]

xxd model.tflite as plain hexdump I get a .txt (size of ~5MB, double the original .tflite file)
[image]
I am parsing this file within the ESP32 and able to load the model, but I would like to be more efficient with my space.

Apologies, I should of specified I am trying to perform a modular type of application with my ESP32 where multiple models can be stored in the SPIFFS filesystem and loaded depending on the use case. My understanding is that the .cc file is embedded in the binary prior to being flashed onto the ESP32. I would not like to flash each time to switch models but develop a way to switch on-demand so the use of file space in my case is important.

khuangCTS, 2022-11-07:
Apologies for the confusion, I am new to TFLite Micro. My understanding is that the model.cc is compiled into my ESP32-S3 and I can have multiple model.cc files also included in the compilation. What I would like to do is have a TFLite micro "container" and provide it with a different model (not necessarily in .cc format, but in its original .tflite format) depending on my use case. 

For example, I can have this "container" grab a model from my ESP32's SPIFFS filesystem and change between them when needed. I particularly do not want to have models compiled into the ESP32 as I believe that the models would be fixed within my ESP32 flash and would require a firmware update(please correct me if I am wrong). I would like to provide an OTA update that will just flash the SPIFFS filesystem as it would not require a whole firmware upgrade to add or remove new models. I hope that make sense. 

I recall you mentioned using the "BuildFromFile" function in TFLite. Could this still be a viable solution?

closed 2023-04-17 by github-actions[bot]