tensorflow/tflite-micro #1527: PR #1512 reintroduced std::round( )

tcal-x, 2022-11-05:
It seems that previously TFLite Micro had avoided `std::round()` -- see comment here: https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/kernels/internal/quantization_util.cc#L396-L398

Is this new use inadvertent, or is it something we (as users) need to deal with now?

Thanks!

tcal-x, 2022-11-05:
The new use: https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/kernels/internal/common.h#L413

Tessil, 2022-11-07:
It is an inadvertent use that was created when https://github.com/tensorflow/tensorflow/blob/v2.10.0/tensorflow/lite/kernels/activations.cc#L122 was moved into `common.h` to replace the previous `gen_lut` for more general usage. 

Sorry for that, it should be safely replaceable by `TfLiteRound`.

advaitjain, 2022-11-16:
I have an internal change that switches std::round to TfLtiteRound in the TF repo. Once that is merged, it will get sync'd to the TFLM repo as well and then we should be good to go.

closed 2022-11-20 by learning-to-play