tensorflow/tflite-micro #1540: Compile with Armclang V6

shixin627, 2022-11-09:
When I try to use armclang v6.18 to compile my RT-Thread project with TensorflowLite package, it couldn't find the files in the path(C:\Keil_v5\ARM\ARMCLANG\include\libcxx) and existed some C++ syntex error(I known this package was writen in c++11).
Can you show any clue for me to figure out these problems. THX!!
[image]
[image]

UlrikHjort, 2022-11-10:
Seems you are compiling with a C compiler and not a C++ compiler

shixin627, 2022-11-10:
You're right. I got the idea to figure out this problem, thank you.

closed 2022-11-10 by shixin627