tensorflow/tflite-micro #1570: RNN tflite model doesn't work when deployed on a microcontroller

jkaldal, 2022-11-17:
**System information**
 * Tensorflow version: 2.9.1

**Describe the feature and the current behavior/state.**
RNN tflite model not working when deployed on a microcontroller unless it is unrolled which increases the size of the model tenfold.

I want to compare different types of RNN tflite-micro models. I have created a custom RNN cell that I want to compare with the LSTM cell, GRU cell, and SimpleRNN cell. 

The following code shows how the network is created and converted to tflite model
```
import tensorflow as tf

# create the rnn cell using one of these RNN cells
units = <value>
#RNNcell = tf.keras.layers.SimpleRNNCell(units)
#RNNcell = tf.keras.layers.GRUCell(units)
RNNcell = tf.keras.layers.LSTMCell(units)

# create the model using the cell in an RNN layer
timesteps = <value>
n_features = <value>
dropout = <value>
dense_units = <value>
outputs = <value>

model = tf.keras.Sequential([
    tf.keras.layers.RNN(RNNcell, input_shape=(timesteps, n_features),
    tf.keras.layers.Dense(dense_units, activation='relu'),
    tf.keras.layers.Dense(outputs, activation='softmax', name="output")
])

# then create model for inference with a fixed batch size
model = tf.keras.Sequential([
    tf.keras.layers.RNN(RNNcell, batch_size=1, input_shape=(timesteps, n_features),
    tf.keras.layers.Dense(dense_units, activation='relu'),
    tf.keras.layers.Dense(outputs, activation='softmax', name="output")
])

#train the model
EPOCHS = <value>
BATCH_SIZE = <value>
LEARNING_RATE = <value>

opt = tf.keras.optimizers.Adam(learning_rate=LEARNING_RATE)
model.compile(optimizer=opt, loss='categorical_crossentropy', metrics=['accuracy'])

model.fit(
    train_X, train_y, 
    epochs=EPOCHS, 
    validation_data=(validation_X, validation_y), 
    shuffle=False, 
    batch_size=BATCH_SIZE)

# move weights to model with fixed batch_size
model_inference.set_weights(loaded_model.get_weights())
model_inference.compile(optimizer=opt, loss='categorical_crossentropy', metrics=['accuracy'])

# save to tflite format
converter = tf.lite.TFLiteConverter.from_keras_model(model_inference)
tfmodel = converter.convert()
open(f'model.tflite', 'wb').write(tfmodel)

## EXTRA
# test operations with tflite interpreter in python
input_data = <value>

interpreter = tf.lite.Interpreter(model_content=tfmodel)
interpreter.allocate_tensors()

_input_index = interpreter.get_input_details()[0]['index']
interpreter.set_tensor(_input_index, input_data)
interpreter.invoke()
```

The tflite model works in the python interpreter but does not run on a microcontroller (nRF5340) unless the RNN layer is unrolled, which increases the size of the model significantly.

This is how the model looks in Netron and the error from the interpreter on the microcontroller indicates that INT32 is not supported for ADD in the WHILE operator.

```
Type INT32 (2) not supported
Node ADD (number 0) failed to invoke with status 1
Node WHILE (number 1) failed to invoke with status 1
```

[image]

**Will this change the current api? How?**
don't know

**Who will benefit with this feature?**
Everybody who needs and thought RNNs were supported in Tensorflow Lite Micro.

**Any Other info.**
Similar to issue #907 but in that issue that person is unrolling. 

Also, a tflite model with created with tf.keras.layers.LSTM creates the operator unidirectionalLSTM but the model is created with a tf.keras.layers.LSTMCell in a tf.keras.layers.RNN layer then it has a reshape and a while operator instead.

Is it possible to use the unidirectionalLSTM operator for the other cells as well somehow?

jkaldal, 2022-11-17:
When I run the analyser on the tflite model:
```
tf.lite.experimental.Analyzer.analyze(model_path='model.tflite')
```
this is what I get
```
=== model.tflite ===

Your TFLite model has '3' subgraph(s). In the subgraph description below,
T# represents the Tensor numbers. For example, in Subgraph#0, the RESHAPE op takes
tensor #0 and tensor #11 as input and produces tensor #13 as output.

Subgraph#0 main(T#0) -> [T#23]
  Op#0 RESHAPE(T#0, T#11) -> [T#13]
  Op#1 WHILE(T#3, T#3, T#1, T#4, T#4, T#13, Cond: Subgraph#1, Body: Subgraph#2) -> [T#14, T#15, T#16, T#17, T#18, T#19]
  Op#2 STRIDED_SLICE(T#16, T#8, T#9, T#10) -> [T#20]
  Op#3 FULLY_CONNECTED(T#20, T#12, T#5) -> [T#21]
  Op#4 FULLY_CONNECTED(T#21, T#7, T#6) -> [T#22]
  Op#5 SOFTMAX(T#22) -> [T#23]

Tensors of Subgraph#0
  T#0(serving_default_lstm_input:0) shape:[1, 128, 6], type:FLOAT32
  T#1(sequential_1/lstm/TensorArrayV2_1) shape:[1, 1, 100], type:FLOAT32 RO 400 bytes
  T#2(sequential_1/lstm/strided_slice_1) shape:[], type:INT32 RO 4 bytes
  T#3(sequential_1/lstm/time) shape:[], type:INT32 RO 4 bytes
  T#4(sequential_1/lstm/zeros) shape:[1, 100], type:FLOAT32 RO 400 bytes
  T#5(sequential_1/dense_1/BiasAdd/ReadVariableOp) shape:[40], type:FLOAT32 RO 160 bytes
  T#6(sequential_1/output/BiasAdd/ReadVariableOp) shape:[6], type:FLOAT32 RO 24 bytes
  T#7(sequential_1/output/MatMul) shape:[6, 40], type:FLOAT32 RO 960 bytes
  T#8(sequential_1/lstm/strided_slice_3) shape:[3], type:INT32 RO 12 bytes
  T#9(sequential_1/lstm/strided_slice_31) shape:[3], type:INT32 RO 12 bytes
  T#10(sequential_1/lstm/strided_slice_32) shape:[3], type:INT32 RO 12 bytes
  T#11(sequential_1/lstm/transpose) shape:[3], type:INT32 RO 12 bytes
  T#12(sequential_1/dense_1/MatMul) shape:[40, 100], type:INT8 RO 4000 bytes
  T#13(sequential_1/lstm/transpose1) shape:[128, 1, 6], type:FLOAT32
  T#14(sequential_1/lstm/while) shape:[], type:INT32
  T#15(sequential_1/lstm/while1) shape:[], type:INT32
  T#16(sequential_1/lstm/while2) shape:[1, 1, 100], type:FLOAT32
  T#17(sequential_1/lstm/while3) shape:[1, 100], type:FLOAT32
  T#18(sequential_1/lstm/while4) shape:[1, 100], type:FLOAT32
  T#19(sequential_1/lstm/while5) shape:[128, 1, 6], type:FLOAT32
  T#20(sequential_1/lstm/strided_slice_33) shape:[1, 100], type:FLOAT32
  T#21(sequential_1/dense_1/MatMul;sequential_1/dense_1/Relu;sequential_1/dense_1/BiasAdd) shape:[1, 40], type:FLOAT32
  T#22(sequential_1/output/MatMul;sequential_1/output/BiasAdd) shape:[1, 6], type:FLOAT32
  T#23(StatefulPartitionedCall:0) shape:[1, 6], type:FLOAT32

Subgraph#1 sequential_1/lstm/while_cond(T#1_0, T#1_1, T#1_2, T#1_3, T#1_4, T#1_5) -> [T#1_7]
  Op#0 LESS(T#1_1, T#1_6) -> [T#1_7]

Tensors of Subgraph#1
  T#1_0(arg0) shape:[], type:INT32
  T#1_1(arg1) shape:[], type:INT32
  T#1_2(arg2) shape:[1, 1, 100], type:FLOAT32
  T#1_3(arg3) shape:[1, 100], type:FLOAT32
  T#1_4(arg4) shape:[1, 100], type:FLOAT32
  T#1_5(arg5) shape:[128, 1, 6], type:FLOAT32
  T#1_6(sequential_1/lstm/strided_slice_11) shape:[], type:INT32 RO 4 bytes
  T#1_7(sequential_1/lstm/while/Less) shape:[], type:BOOL

Subgraph#2 sequential_1/lstm/while_body(T#2_0, T#2_1, T#2_2, T#2_3, T#2_4, T#2_5) -> [T#2_13, T#2_11, T#2_31, T#2_30, T#2_28, T#2_5]
  Op#0 ADD(T#2_1, T#2_6) -> [T#2_11]
  Op#1 FULLY_CONNECTED(T#2_3, T#2_10, T#-1) -> [T#2_12]
  Op#2 ADD(T#2_0, T#2_6) -> [T#2_13]
  Op#3 GATHER(T#2_5, T#2_1) -> [T#2_14]
  Op#4 FULLY_CONNECTED(T#2_14, T#2_9, T#-1) -> [T#2_15]
  Op#5 ADD(T#2_15, T#2_12) -> [T#2_16]
  Op#6 ADD(T#2_16, T#2_7) -> [T#2_17]
  Op#7 SPLIT(T#2_6, T#2_17) -> [T#2_18, T#2_19, T#2_20, T#2_21]
  Op#8 LOGISTIC(T#2_18) -> [T#2_22]
  Op#9 LOGISTIC(T#2_19) -> [T#2_23]
[... 43 lines cut ...]
---------------------------------------------------------------
Your TFLite model has ‘1’ signature_def(s).

Signature#0 key: 'serving_default'
- Subgraph: Subgraph#0
- Inputs: 
    'lstm_input' : T#0
- Outputs: 
    'output' : T#23

---------------------------------------------------------------
              Model size:      58128 bytes
    Non-data buffer size:       8004 bytes (13.77 %)
  Total data buffer size:      50124 bytes (86.23 %)
          - Subgraph#0  :       6000 bytes (10.32 %)
          - Subgraph#1  :          4 bytes (00.01 %)
          - Subgraph#2  :      44016 bytes (75.72 %)
    (Zero value buffers):        804 bytes (01.38 %)

* Buffers of TFLite model are mostly used for constant tensors.
  And zero value buffers are buffers filled with zeros.
  Non-data buffers area are used to store operators, subgraphs and etc.
  You can find more details from https://github.com/tensorflow/tensorflow/blob/master/tensorflow/lite/schema/schema.fbs

```

closed 2023-04-06 by github-actions[bot]

TayyabaZainab0807, 2023-06-26:
did you find the solution for this?

jkaldal, 2023-06-26:
You have to unroll the network

jonnor, 2023-07-12:
How does one unroll the network?

jkaldal, 2023-07-12:
It is an argument in tf.keras.layers.RNN

Not a good solution for large networks.