tensorflow/tflite-micro #1609: Unable to create examples due to "flatbuffers/flatbuffers.h' file not found"

subrockmann, 2022-12-03:
I am trying to create the "hello_world" example with the following code:
```
make -f tensorflow/lite/micro/tools/project_generation/Makefile hello_world
```

Unfortunately, this results in a fatal error: 
```'flatbuffers/flatbuffers.h' file not found #include "flatbuffers/flatbuffers.h"
```
Running on WSL2 Ubuntu 20.04 on Windows 10.

Looking at the directory structure, this file is not there. Therefore it seems that the Makefile needs to be updated.

tanmaydas82, 2022-12-03:
Please try the following command to build the hello world example
```
make -f tensorflow/lite/micro/tools/make/Makefile hello_world
```

subrockmann, 2022-12-04:
Thank you @tanmaydas82 : I have tried your suggestion.

```
make -f tensorflow/lite/micro/tools/make/Makefile hello_world
```

However this creates another error:

```
make: tensorflow/lite/micro/tools/make/flatbuffers_download.sh: Command not found
tensorflow/lite/micro/tools/make/Makefile:489: *** Something went wrong with the flatbuffers download: .  Stop.
```

tanmaydas82, 2022-12-05:
The reason of this failure can be an out of date `flatbuffer` download in the following directory 
`tensorflow/lite/micro/tools/make/downloads/`

Please try to run the make after deleting the `tensorflow/lite/micro/tools/make/downloads/flatbuffers` directory completely.

subrockmann, 2022-12-05:
@tanmaydas82 : The downloads folder is empty. There is no ``` flatbuffers```  directory inside.

I have therefore decided to pull the latest version from git (my original code was about a week old) and run again:

```
make -f tensorflow/lite/micro/tools/make/Makefile hello_world
```
I get a new error:

```
File "/mnt/i/tinyml/tflite-micro/tensorflow/lite/micro/tools/generate_cc_arrays.py", line 26, in <module>
    from PIL import Image
ModuleNotFoundError: No module named 'PIL'
```

I did not find any requirements.txt file that is mentioning this requirement. Am I missing something here?

Thanks for your support.

tanmaydas82, 2022-12-08:
@subrockmann : Please share the full log, if possible. I can then look at it more closely and potentially we can find a solution.  Thank you!

subrockmann, 2022-12-10:
@tanmaydas82 : I ran the code again and I have received the following output:

```
(base) dev@HP-Omen:/mnt/i/tinyml/tflite-micro$ make -f tensorflow/lite/micro/tools/make/Makefile hello_world
tensorflow/lite/micro/tools/make/downloads/flatbuffers already exists, skipping the download.
tensorflow/lite/micro/tools/make/downloads/kissfft already exists, skipping the download.
tensorflow/lite/micro/tools/make/downloads/pigweed already exists, skipping the download.
Traceback (most recent call last):
  File "/mnt/i/tinyml/tflite-micro/tensorflow/lite/micro/tools/generate_cc_arrays.py", line 26, in <module>
    from PIL import Image
ModuleNotFoundError: No module named 'PIL'
tensorflow/lite/micro/examples/hello_world/Makefile.inc:38: *** Something went wrong: .  Stop.
```

After that I removed the folders "flatbuffers", "kissfft" and "pigweed" from the downloads folder and ran again the following command:

""" 
(base) dev@HP-Omen:/mnt/i/tinyml/tflite-micro$ make -f tensorflow/lite/micro/tools/make/Makefile hello_world
--2022-12-10 10:21:25--  https://github.com/google/flatbuffers/archive/a66de58af9565586832c276fbb4251fc416bf07f.zip
Resolving github.com (github.com)... 140.82.121.4
Connecting to github.com (github.com)|140.82.121.4|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://codeload.github.com/google/flatbuffers/zip/a66de58af9565586832c276fbb4251fc416bf07f [following]
--2022-12-10 10:21:25--  https://codeload.github.com/google/flatbuffers/zip/a66de58af9565586832c276fbb4251fc416bf07f
Resolving codeload.github.com (codeload.github.com)... 140.82.121.9
Connecting to codeload.github.com (codeload.github.com)|140.82.121.9|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [application/zip]
Saving to: ‘/tmp/tmp.94DieUMG9r/a66de58af9565586832c276fbb4251fc416bf07f.zip’

/tmp/tmp.94DieUMG9r/a66de58af     [      <=>                                         ]   2.74M  2.47MB/s    in 1.1s

2022-12-10 10:21:27 (2.47 MB/s) - ‘/tmp/tmp.94DieUMG9r/a66de58af9565586832c276fbb4251fc416bf07f.zip’ saved [2872508]

Switched to a new branch 'tflm'
Applying /mnt/i/tinyml/tflite-micro/tensorflow/lite/micro/tools/make/downloads/flatbuffers/.//../../flatbuffers.patch to /mnt/i/tinyml/tflite-micro/tensorflow/lite/micro/tools/make/downloads/flatbuffers/./
--2022-12-10 10:22:11--  https://github.com/mborgerding/kissfft/archive/refs/tags/v130.zip
Resolving github.com (github.com)... 140.82.121.4
Connecting to github.com (github.com)|140.82.121.4|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://codeload.github.com/mborgerding/kissfft/zip/refs/tags/v130 [following]
--2022-12-10 10:22:11--  https://codeload.github.com/mborgerding/kissfft/zip/refs/tags/v130
Resolving codeload.github.com (codeload.github.com)... 140.82.121.9
Connecting to codeload.github.com (codeload.github.com)|140.82.121.9|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [application/zip]
Saving to: ‘/tmp/tmp.uTn4GzCxei/v130.zip’

/tmp/tmp.uTn4GzCxei/v130.zip      [ <=>                                              ]  59.09K  --.-KB/s    in 0.05s

2022-12-10 10:22:11 (1.24 MB/s) - ‘/tmp/tmp.uTn4GzCxei/v130.zip’ saved [60513]

Switched to a new branch 'tflm'
Applying /mnt/i/tinyml/tflite-micro/tensorflow/lite/micro/tools/make/downloads/kissfft/.//../../kissfft.patch to /mnt/i/tinyml/tflite-micro/tensorflow/lite/micro/tools/make/downloads/kissfft/./
Cloning into 'tensorflow/lite/micro/tools/make/downloads/pigweed'...
remote: Sending approximately 37.74 MiB ...
remote: Counting objects: 34, done
remote: Finding sources: 100% (34/34)
remote: Total 56158 (delta 26956), reused 56144 (delta 26956)
Receiving objects: 100% (56158/56158), 37.71 MiB | 3.86 MiB/s, done.
Resolving deltas: 100% (26956/26956), done.
Updating files: 100% (3522/3522), done.
Updating files: 100% (3570/3570), done.
Note: switching to '47268dff45019863e20438ca3746c6c62df6ef09'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at 47268dff pw_hdlc_lite: Client I/O improvements
Switched to a new branch 'tflm'
Applying /mnt/i/tinyml/tflite-micro/tensorflow/lite/micro/tools/make/downloads/pigweed/.//../../pigweed.patch to /mnt/i/tinyml/tflite-micro/tensorflow/lite/micro/tools/make/downloads/pigweed/./
warning: pw_presubmit/py/pw_presubmit/format_code.py has type 100644, expected 100755
warning: pw_presubmit/py/pw_presubmit/pigweed_presubmit.py has type 100644, expected 100755
Traceback (most recent call last):
  File "/mnt/i/tinyml/tflite-micro/tensorflow/lite/micro/tools/generate_cc_arrays.py", line 26, in <module>
    from PIL import Image
ModuleNotFoundError: No module named 'PIL'
tensorflow/lite/micro/examples/hello_world/Makefile.inc:38: *** Something went wrong: .  Stop.
(base) dev@HP-Omen:/mnt/i/tinyml/tflite-micro$
 """

It is still the same issue that requires the installation of the PIL module.

advaitjain, 2022-12-10:
Please see https://github.com/tensorflow/tflite-micro/blob/main/third_party/requirements.txt for additional pip packages that are needed.

PIL is part of the pillow pip package.

subrockmann, 2022-12-26:
@advaitjain: Thanks that has soved the issue. I did not expect to find a requirements.txt that deep into the project structure.

closed 2023-02-08 by rkuester