tensorflow/tflite-micro #1615: On WSL, I am unable to make a project using the arm toolchain.

phuoctan4141, 2022-12-09:
I can't complete this project with the current version. However, on the old version from two months ago, this project could be successful.
[image]
[image]

mansnils, 2022-12-12:
Hi @phuoctan4141 
Thanks for reporting this, we will look into it.

mansnils, 2022-12-12:
Hi @phuoctan4141 

Did you mean TARGET=cortex_m_corstone_300 instead TARGET=cortex_m_generic?

The TARGET=cortex_m_generic is just meant to build the micro lib and then link this to the application outside of TFLM.
So it has no support for building any unit test or example test within TFLM.
For that the TARGET=cortex_m_corstone_300 could possible be used instead.

So I don't think this has ever worked:
make -j 4 -f tensorflow/lite/micro/tools/make/Makefile OPTIMIZED_KERNEL_DIR=cmsis_nn TARGET=cortex_m_generic TARGET_ARCH=cortex-m0plus  hello_world

You can also see the tests for these targets:

https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/tools/ci_build/test_cortex_m_corstone_300.sh
https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/tools/ci_build/test_cortex_m_generic.sh

mansnils, 2022-12-13:
Could you elaborate on this project?

It looks like you are using CMake. The internal examples in TFLM does not use CMake.

phuoctan4141, 2022-12-13:
I could fix all program. Thanks for your support.

closed 2022-12-13 by phuoctan4141