tensorflow/tflite-micro #1621: The tlfm-micro-tree file generated under linux, was incorrectly compiled in VS2019

lzlwakeup, 2022-12-15:
Hi,
My idea is to generate a cross-platform source file and header file for compilation on different platforms. According to the document description, using 
"python3 tensorflow/lite/micro/tools/project_generation/create_tflm_tree.py" 
method to generate the corresponding files, and then write a test file to compile validation.
 if it can be done, The cross-platform dependency file I generated is correct. 

However, in the test, the compilation error.

_Question 1_

**LNK2001	External symbols that cannot be resolved"enum TfLiteStatus __cdecl tflite::CalculateActivationRangeQuantized(struct TfLiteContext *,enum TfLiteFusedActivation,struct TfLiteTensor *,int *,int *)" (?CalculateActivationRangeQuantized@tflite@@YA?AW4TfLiteStatus@@PAUTfLiteContext@@W4TfLiteFusedActivation@@PAUTfLiteTensor@@PAH3@Z)	tflite_pro	D:\workspace\tflite\tflite_pro\fully_connected_common.obj	1	

External symbols that cannot be resolved "enum TfLiteStatus __cdecl tflite::CalculateActivationRangeQuantized(struct TfLiteContext *,enum TfLiteFusedActivation,struct TfLiteTensor *,int *,int *)" (?CalculateActivationRangeQuantized@tflite@@YA?AW4TfLiteStatus@@PAUTfLiteContext@@W4TfLiteFusedActivation@@PAUTfLiteTensor@@PAH3@Z)	tflite_pro	D:\workspace\tflite\tflite_pro\mul_common.obj	1	

External symbols that cannot be resolved "enum TfLiteStatus __cdecl tflite::CalculateActivationRangeQuantized(struct TfLiteContext *,enum TfLiteFusedActivation,struct TfLiteTensor *,int *,int *)" (?CalculateActivationRangeQuantized@tflite@@YA?AW4TfLiteStatus@@PAUTfLiteContext@@W4TfLiteFusedActivation@@PAUTfLiteTensor@@PAH3@Z)	tflite_pro	D:\workspace\tflite\tflite_pro\pooling_common.obj	1	

External symbols that cannot be resolved "enum TfLiteStatus __cdecl tflite::CalculateActivationRangeQuantized(struct TfLiteContext *,enum TfLiteFusedActivation,struct TfLiteTensor *,int *,int *)" (?CalculateActivationRangeQuantized@tflite@@YA?AW4TfLiteStatus@@PAUTfLiteContext@@W4TfLiteFusedActivation@@PAUTfLiteTensor@@PAH3@Z)	tflite_pro	D:\workspace\tflite\tflite_pro\sub_common.obj	1	

External symbols that cannot be resolved "enum TfLiteStatus __cdecl tflite::CalculateActivationRangeQuantized(struct TfLiteContext *,enum TfLiteFusedActivation,struct TfLiteTensor *,int *,int *)" (?CalculateActivationRangeQuantized@tflite@@YA?AW4TfLiteStatus@@PAUTfLiteContext@@W4TfLiteFusedActivation@@PAUTfLiteTensor@@PAH3@Z)	tflite_pro	D:\workspace\tflite\tflite_pro\add_common.obj	1	

External symbols that cannot be resolved "enum TfLiteStatus __cdecl tflite::CalculateActivationRangeQuantized(struct TfLiteContext *,enum TfLiteFusedActivation,struct TfLiteTensor *,int *,int *)" (?CalculateActivationRangeQuantized@tflite@@YA?AW4TfLiteStatus@@PAUTfLiteContext@@W4TfLiteFusedActivation@@PAUTfLiteTensor@@PAH3@Z)	tflite_pro	D:\workspace\tflite\tflite_pro\add_n.obj	1	
...
...
External symbols that cannot be resolved "void __cdecl tflite::tensor_utils::MatrixScalarMultiplyAccumulate(signed char const *,int,int,int,int *)" (?MatrixScalarMultiplyAccumulate@tensor_utils@tflite@@YAXPBCHHHPAH@Z)，function "enum TfLiteStatus __cdecl tflite::`anonymous namespace'::PrecomputeZeroPointTimesWeightWithBias(struct TfLiteContext *,int,struct TfLiteTensor const *,struct TfLiteTensor const *,int * *)" (?PrecomputeZeroPointTimesWeightWithBias@?A0xb72b3bf6@tflite@@YA?AW4TfLiteStatus@@PAUTfLiteContext@@HPBUTfLiteTensor@@1PAPAH@Z) The symbol is referenced	tflite_pro	D:\workspace\tflite\tflite_pro\unidirectional_sequence_lstm.obj	1**		

too many do not show all, all of them is "External symbols that cannot be resolved".

_Question 2_

In file ..\tensorflow\lite\micro\test_helper.cc, 

constexpr size_t inputs_size = 0;
const int32_t inputs[inputs_size] = {};
constexpr size_t operator_inputs_size = 0;
const int32_t operator_inputs[operator_inputs_size] = {};

An error message indicating that an array of length 0 cannot be assigned, The array must be greater than 0.

Looking forward to your reply!
Thanks!

lzlwakeup, 2022-12-15:
All source file paths are added to the compiler's links

lzlwakeup, 2022-12-15:
Or there are quick ways to separate files across platforms. Maybe someone else has been through this and can provide a reference.

sharvin-vittappan, 2023-01-02:
Question 1: There are files with same name in the tflite  micro repo that you are comparing (eg: kernel_utils.cc), when you compile them in MSVC, only one of them will be considered. 
I do not know the exact solution, but renaming one of those files can be a work around.