tensorflow/tflite-micro #1622: Flatten kernel namespace: ARG_MIN_MAX

ddavis-2015, 2022-12-15:
@tensorflow/micro 

Flatten the C++ namespace of all kernel operators as per [abseil / Tip of the Week #130: Namespace Naming](https://abseil.io/tips/130)

This issue pertains to: ARG_MIN_MAX

kslavka, 2022-12-19:
This commit fails the build of test_hello_world_test:
make -f tensorflow/lite/micro/tools/make/Makefile test_hello_world_test
all_ops_resolver.cc:(.text._ZN6tflite14AllOpsResolverC2Ev+0xa1): undefined reference to `tflite::ops::micro::Register_ARG_MAX()'
/usr/bin/ld: all_ops_resolver.cc:(.text._ZN6tflite14AllOpsResolverC2Ev+0xc0): undefined reference to `tflite::ops::micro::Register_ARG_MIN()'
/usr/bin/ld: all_ops_resolver.cc:(.text._ZN6tflite14AllOpsResolverC2Ev+0x758): undefined reference to `tflite::ops::micro::Register_PAD()'
/usr/bin/ld: all_ops_resolver.cc:(.text._ZN6tflite14AllOpsResolverC2Ev+0x777): undefined reference to `tflite::ops::micro::Register_PADV2()'

advaitjain, 2022-12-19:
Can you please try a `make clean` and then rebuild? The TFLM Makefile doesn't do a good job with with incremental builds and that might be the issue here.

kslavka, 2022-12-20:
Clean did the job. Thanks
I tried to clean and build before I posted but probably something messed up in the clean.

closed 2022-12-22 by ddavis-2015