tensorflow/tflite-micro #1663: make test_ hello_ world_ The compiler reports an error during test

TomMichey, 2023-01-14:
Error reported when compiling helloworld，“error: assuming signed overflow does not occur when assuming that (X - c) > X is always false [-Werror=strict-overflow]”，what should I do？
`make -f tensorflow/lite/micro/tools/make/Makefile test_hello_world_test
tensorflow/lite/micro/tools/make/downloads/flatbuffers already exists, skipping the download.
tensorflow/lite/micro/tools/make/downloads/kissfft already exists, skipping the download.
tensorflow/lite/micro/tools/make/downloads/pigweed already exists, skipping the download.
find: ‘tensorflow/lite/micro/tools/gen_micro_mutable_op_resolver_test/person_detect’: No such file or directory
g++ -std=c++11 -fno-rtti -fno-exceptions -fno-threadsafe-statics -Wnon-virtual-dtor -Werror -fno-unwind-tables -ffunction-sections -fdata-sections -fmessage-length=0 -DTF_LITE_STATIC_MEMORY -DTF_LITE_DISABLE_X86_NEON -Wsign-compare -Wdouble-promotion -Wshadow -Wunused-variable -Wunused-function -Wswitch -Wvla -Wall -Wextra -Wmissing-field-initializers -Wstrict-aliasing -Wno-unused-parameter  -DTF_LITE_USE_CTIME -Os -I. -Itensorflow/lite/micro/tools/make/downloads/gemmlowp -Itensorflow/lite/micro/tools/make/downloads/flatbuffers/include -Itensorflow/lite/micro/tools/make/downloads/ruy -Igen/linux_x86_64_default/genfiles/ -Igen/linux_x86_64_default/genfiles/ -Itensorflow/lite/micro/tools/make/downloads/kissfft -c gen/linux_x86_64_default/genfiles/tensorflow/lite/micro/examples/hello_world/hello_world_model_data.cc -o gen/linux_x86_64_default/obj/core/gen/linux_x86_64_default/genfiles/tensorflow/lite/micro/examples/hello_world/hello_world_model_data.o
g++ -std=c++11 -fno-rtti -fno-exceptions -fno-threadsafe-statics -Wnon-virtual-dtor -Werror -fno-unwind-tables -ffunction-sections -fdata-sections -fmessage-length=0 -DTF_LITE_STATIC_MEMORY -DTF_LITE_DISABLE_X86_NEON -Wsign-compare -Wdouble-promotion -Wshadow -Wunused-variable -Wunused-function -Wswitch -Wvla -Wall -Wextra -Wmissing-field-initializers -Wstrict-aliasing -Wno-unused-parameter  -DTF_LITE_USE_CTIME -O2 -I. -Itensorflow/lite/micro/tools/make/downloads/gemmlowp -Itensorflow/lite/micro/tools/make/downloads/flatbuffers/include -Itensorflow/lite/micro/tools/make/downloads/ruy -Igen/linux_x86_64_default/genfiles/ -Igen/linux_x86_64_default/genfiles/ -Itensorflow/lite/micro/tools/make/downloads/kissfft -c tensorflow/lite/micro/kernels/fully_connected.cc -o gen/linux_x86_64_default/obj/kernels/tensorflow/lite/micro/kernels/fully_connected.o
tensorflow/lite/micro/kernels/fully_connected.cc: In function ‘void tflite::reference_integer_ops::FullyConnected(const tflite::FullyConnectedParams&, const tflite::RuntimeShape&, const int16_t*, const tflite::RuntimeShape&, const int8_t*, const tflite::RuntimeShape&, const AccumScalar*, const tflite::RuntimeShape&, int16_t*) [with AccumScalar = long int; int16_t = short int; int8_t = signed char]’:
tensorflow/lite/micro/kernels/fully_connected.cc:192:1: error: assuming signed overflow does not occur when assuming that (X - c) > X is always false [-Werror=strict-overflow]
 }  // namespace tflite
 ^
tensorflow/lite/micro/kernels/fully_connected.cc: In function ‘void tflite::reference_integer_ops::FullyConnected(const tflite::FullyConnectedParams&, const tflite::RuntimeShape&, const int8_t*, const tflite::RuntimeShape&, const int8_t*, const tflite::RuntimeShape&, const int32_t*, const tflite::RuntimeShape&, int8_t*)’:
tensorflow/lite/micro/kernels/fully_connected.cc:192:1: error: assuming signed overflow does not occur when assuming that (X - c) > X is always false [-Werror=strict-overflow]
In file included from ./tensorflow/lite/kernels/internal/types.h:24:0,
                 from ./tensorflow/lite/micro/kernels/fully_connected.h:22,
                 from tensorflow/lite/micro/kernels/fully_connected.cc:16:
./tensorflow/lite/kernels/internal/runtime_shape.h: In function ‘TfLiteStatus tflite::{anonymous}::Eval(TfLiteContext*, TfLiteNode*)’:
./tensorflow/lite/kernels/internal/runtime_shape.h:61:31: error: assuming signed overflow does not occur when assuming that (X - c) > X is always false [-Werror=strict-overflow]
     TFLITE_DCHECK_LT(i, size_);
                               ^
./tensorflow/lite/kernels/internal/runtime_shape.h:61:31: error: assuming signed overflow does not occur when assuming that (X - c) > X is always false [-Werror=strict-overflow]
     TFLITE_DCHECK_LT(i, size_);
                               ^
./tensorflow/lite/kernels/internal/runtime_shape.h:61:31: error: assuming signed overflow does not occur when assuming that (X - c) > X is always false [-Werror=strict-overflow]
     TFLITE_DCHECK_LT(i, size_);
                               ^
tensorflow/lite/micro/kernels/fully_connected.cc:79:14: error: assuming signed overflow does not occur when assuming that (X - c) > X is always false [-Werror=strict-overflow]
 TfLiteStatus Eval(TfLiteContext* context, TfLiteNode* node) {
              ^
tensorflow/lite/micro/kernels/fully_connected.cc:79:14: error: assuming signed overflow does not occur when assuming that (X - c) > X is always false [-Werror=strict-overflow]
In file included from ./tensorflow/lite/kernels/internal/types.h:24:0,
                 from ./tensorflow/lite/micro/kernels/fully_connected.h:22,
                 from tensorflow/lite/micro/kernels/fully_connected.cc:16:
./tensorflow/lite/kernels/internal/runtime_shape.h:61:31: error: assuming signed overflow does not occur when assuming that (X - c) > X is always false [-Werror=strict-overflow]
     TFLITE_DCHECK_LT(i, size_);
                               ^
cc1plus: all warnings being treated as errors
tensorflow/lite/micro/tools/make/Makefile:686: recipe for target 'gen/linux_x86_64_default/obj/kernels/tensorflow/lite/micro/kernels/fully_connected.o' failed
make: *** [gen/linux_x86_64_default/obj/kernels/tensorflow/lite/micro/kernels/fully_connected.o] Error 1
`

subrockmann, 2023-01-17:
@TomMichey: Your second question is related to the problem in this currently unsolved [issue #1644](https://github.com/tensorflow/tflite-micro/issues/1644). 

‘tensorflow/lite/micro/tools/gen_micro_mutable_op_resolver_test/person_detect’: No such file or directory

TomMichey, 2023-01-18:
> @TomMichey: Your second question is related to the problem in this currently unsolved [issue #1644](https://github.com/tensorflow/tflite-micro/issues/1644).
> 
> ‘tensorflow/lite/micro/tools/gen_micro_mutable_op_resolver_test/person_detect’: No such file or directory

Although this warning "person_detect" is not related to my "helloworld", I want to know whether you can run your program after skipping this warning. thank you

TomMichey, 2023-01-18:
> 

Although this warning "person_detect" is not related to my "helloworld", I want to know whether you can run your program after skipping this warning. thank you！

TomMichey, 2023-01-30:
help

rkuester, 2023-02-08:
Hi @TomMichey, on which gcc version and platform are you seeing this? I'm unable to reproduce it with the two latest gcc lines, 10.4.0 or 12.2.0, on x86_64 Debian.

TomMichey, 2023-02-14:
> Hi @TomMichey, on which gcc version and platform are you seeing this? I'm unable to reproduce it with the two latest gcc lines, 10.4.0 or 12.2.0, on x86_64 Debian.
It was “gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.12)”. I failed to compile.