tensorflow/tflite-micro #1687: sparkfun_edge_makefile.inc: No such file or directory

magarita2022, 2023-01-27:
Hi,
I clone the repo and run the make command: make -f tensorflow/lite/micro/tools/make/Makefile TARGET=sparkfun_edge micro_speech_bin
But it returns an error:
tensorflow/lite/micro/tools/make/Makefile:524: tensorflow/lite/micro/tools/make/targets/sparkfun_edge_makefile.inc: No such file or directory
make: *** No rule to make target 'tensorflow/lite/micro/tools/make/targets/sparkfun_edge_makefile.inc'.  Stop.
Who can give me some help please?

ktorimaru, 2023-01-31:
Same problem! Where can I get the file?

Uniqueml, 2023-02-10:
So, In order to do the make command, make sure you are in the tflite micro root instead of the tensorflow root. If you dont have the tensorflow lite micro root, then you need to get it [here](https://github.com/tensorflow/tflite-micro). Then you navigate to the same file path given above and you will need install a few dependencies: Bzip2, zip, unzip,  GCC, G++, Python 3, pip, image,  

additionally, you will need: 
git clone tensorflow/tflite-micro tflite-micro
apt install gcc
apt install g++
apt install unzip
apt install zip
apt install python3
apt install python3-pip
apt install bzip2
pip install image

then you need to run [this](https://github.com/advaitjain/tflite-micro-sparkfun-edge-examples/blob/0cc8217949d3069d32187c2f2bb62aaedd73c669/scripts/sync_from_tflite_micro.sh#L55) if you are using an edge board.
make -f tensorflow/lite/micro/tools/make/Makefile third_party_downloads

make sure all your installations are inside the ubuntu wsl or if you're on linux don't worry about that part. 

Then once you've verified all the above are installed come back and let me know if that fixed it. 

If you just want the folder though here go, I'm fairly certain you'll still need the dependencies to do anything with it, though I could be wrong.
[micro_speech.zip](https://github.com/tensorflow/tflite-micro/files/10705384/micro_speech.zip)
u go.

advaitjain, 2023-02-10:
Individual hardware targets are no longer supported from the TFLM repository. Please see https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/docs/new_platform_support.md if you'd like to add support for your board.

Current community supported TFLM examples are listed at https://github.com/tensorflow/tflite-micro#community-supported-tflm-examples

Note that Sparkfun Edge is available for demonstration purposes only: https://github.com/advaitjain/tflite-micro-sparkfun-edge-examples

closed 2023-02-10 by advaitjain