tensorflow/tflite-micro #1720: Issues with tflite-micro integration as a static library

usmnzain, 2023-02-08:
I am integrating tflite micro as a standalone project. I use cmake to build the entire project (using tflm as a static library). I got compiler errors 
1. In `micro_allocator.cc`: `error: 'static void tflite::GreedyMemoryPlanner::operator delete(void*)' is private within this context` 
2. In `micro_error_reporter.cc`: `error: 'static void tflite::MicroErrorReporter::operator delete(void*)' is private within this context`

I fixed these errors by moving the `TF_LITE_REMOVE_VIRTUAL_DELETE` to public in the corresponding header files. I tried it first by adding the compiler flags mentioned in this issue https://github.com/tensorflow/tflite-micro/issues/483 but that didn't solve it.

mansnils, 2023-02-10:
Hi @usmnzain,
Perhaps this might help: https://review.mlplatform.org/plugins/gitiles/ml/ethos-u/ethos-u-core-software/+/refs/heads/master/tflite_micro.cmake

usmnzain, 2023-02-13:
I only face the errors when I generate project with `python3 tensorflow/lite/micro/tools/project_generation/create_tflm_tree.py` and build it as a static library to support my custom platform. Building the tflite-micro hello world example does not cause any problems as I was able to run the binary on spike.

closed 2023-03-27 by github-actions[bot]

Ruairi-Grant, 2023-11-16:
> I only face the errors when I generate project with `python3 tensorflow/lite/micro/tools/project_generation/create_tflm_tree.py` and build it as a static library to support my custom platform. [...]

Did you ever get a soution for this?

gamelaster, 2025-02-08:
For @Ruairi-Grant and for others, I solved it by using `-fno-rtti;-fno-exceptions;-fno-threadsafe-statics` for CXX files.