tensorflow/tflite-micro #1762: microprofiler.LogCsv() does not return ticks

subrockmann, 2023-02-18:
There is a bug in [micro_profiler.cc](https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/micro_profiler.cc) and using microprofiler.LogCsv() does not return the ticks value.

This line of code:
```
MicroPrintf("%d,%s,%" PRIu32, i, tags_[i], ticks);
```
should be replaced by this code:

```
MicroPrintf("%d,%s,d%" PRIu32, i, tags_[i], ticks);
```

ddavis-2015, 2023-02-19:
Thank you for your feedback regarding the TFLM codebase.  I am afraid we cannot implement your suggested change.

The definition of `PRIu32` is a string constant defined in `<cinttypes>` which is part of the standard C++11 library.  As such it is concatenated, by the compiler, with any and all adjacent strings.

The value of `PRIu32` should be `"u"`.  If you are using a plaltform on which this is not the case, we would like to know more about this.  Please provide platform details including processor variant, the toolchain being used, toolchain version, and the value to which `PRIu32` resolves.

Please feel free to reopen this issue if you wish to add additional information as requested above.

closed 2023-02-19 by ddavis-2015

subrockmann, 2023-02-21:
@ddavis-2015: I am not relating to the definition of "Priu32" but to the fact that the code is missing a "d" (after the third % in the  print statement) in:

```
MicroPrintf("%d,%s,%" PRIu32, i, tags_[i], ticks);
```
and therefore just returns the unit but not the associated value. Sorry for the typo in my above presented code.