tensorflow/tflite-micro #1784: Encounter segment error when  trying to deploy tflite model

HJL8055, 2023-02-23:
Hi,
I encountered segment error when I tried to deploy my tflite model.
Then, I found that error appeared while "AllocateTensors()"->"graph_.PrepareSubgraphs()"->"registration->prepare(context_, node)"
There, I printed the node id and registration->builtin_code.
Result list below.
[image]
After using netron, the node 24 correspond to FullyConnected.
[image]
If I remove the AddFullyConnected() for micro_op_resolver, it shows the version of FullyConnected is verson'5'.
[image]
I am not sure whether some preblems occur here.

HJL8055, 2023-03-01:
Recently, I find that the error is raised by  the following code. 
TFLITE_DCHECK(filter->params.zero_point == 0);
This code restricts the zeropoints of the weight in fullyconnected-layer should always be zero.
[image]
But mine is -10 .
I wonder why my zeropoint is not zero while I use tensorflow-Post-training quantization-Full integer quantization to get my quantization_weight and it's hard for me to find the configure to restricts the zeropoints.

closed 2023-03-31 by github-actions[bot]

karkapur, 2024-07-15:
Hi,

Has this error been addressed? I'm getting the same error and was wondering if someone has been able to figure this out.