tensorflow/tflite-micro #1823: interpreter cannot load model correctly (xtensa)

happyEmart, 2023-03-14:
Hello,

I'm trying to test micro speech test(/tensorflow/lite/micro/examples/micro_speech).
It works fine when I do test in linux with below command :

make -f tensorflow/lite/micro/tools/make/Makefile TARGET=xtensa OPTIMIZED_KERNEL_DIR=xtensa TARGET_ARCH=hifi4 XTENSA_TOOLS_VERSION=RI-2021.8-linux XTENSA_CORE=hifi4_ss_spfpu_7 test_micro_speech_test

But in xplorer of xtensa, it failed after interpreter.AllocateTensors();
It is suppose to load model with input type int8 but it always assigned as float32 and because of incorrect model loading
the test eventually failed with exception. 

I used same source code:
micro_speech_test.cc
https://github.com/tensorflow/tflite-micro/tree/main/tensorflow/lite/micro/examples/micro_speech

Does anybody know about this issue and how to resolve this?

happyEmart, 2023-04-10:
Hi HiFi users,

I resolved this issue with additional compile symbol 'TF_LITE_STATIC_MEMORY'.
you can add this as addl compiler '-DTF_LITE_STATIC_MEMORY' for xt-run command.

If you made other class to call micro interpreter you must assign tensor arena array as static.
Otherwise, you would get out of bound exception.

:)

closed 2023-04-10 by happyEmart

happyEmart, 2023-05-21:
you might need to use tensor_arena buffer as static if you separate functions.

oniigiirii, 2023-05-21:
Declaring the tensor arena buffer as static does not solve the issue.