tensorflow/tflite-micro #1825: Type INT32 (2) not supported for LSTM Models

ts532, 2023-03-14:
I've been trying to run an LSTM Model on an ESP 32 (Adafruit HUZZAH 32 - ESP32 Feather Board), however have been stuck at the following issue. This has been posted to the tflite-micro-esp-examples github page [here](https://github.com/espressif/tflite-micro-esp-examples/issues/32), however I was suggested to also post it here as a more general issue.

Essentially, when calling _interpreter->invoke();_ the following error occurs:
```
Type INT32 (2) not supported.
Node ADD (number 4) failed to invoke with status 1
Node WHILE (number 2) failed to invoke with status 1
```
The model has been converted to at least have int8 inputs and outputs, however it appears that at some point it gets converted to INT32 and crashes. I have tried to modify add.cc to add the ability to use INT32, however this results in the program instead crashing.

**Model Process**
The model being used is an LSTM Model expecting a 20x8 input (20 samples for 8 sensors) and producing a 1x10 output (10 future prediction steps) that looks like the following:
```
model = tf.keras.Sequential([
    Input(shape=(20,8), name="Input"),
    LSTM(units=48, return_sequences=True, activation='relu', unroll=False), 
    LSTM(units=16, return_sequences=False, activation='relu',unroll=False),
    layers.Flatten(),
    Dense(units=pred, name="output")
    ])
```
This is then quantised to int8 format:
```
def representative_data_gen():
    for i in range(300): #for arbitrary loops
        yield[tf.random.uniform(shape=[1,20,8], minval=-1, maxval=1.)] #yield random values between -1 and 1 of appropriate input size

converter = tf.lite.TFLiteConverter.from_keras_model(model)
converter.optimizations = [tf.lite.Optimize.DEFAULT]
converter.representative_dataset = representative_data_gen #"Representative Dataset" required to quantise to int8
converter.target_spec.supported_ops = [tf.lite.OpsSet.TFLITE_BUILTINS_INT8]
converter.inference_input_type = tf.int8  
converter.inference_output_type = tf.int8  
tflite_model = converter.convert()
open(fileLoc+fileName, "wb").write(tflite_model)
```
Finally, it is converted from .tflite format to .h format with xxd + vim.

**ESP32 Code**
As stated, this process results in the int32 error when this model is run in the microprocessor.
```
#include <Arduino.h>
#include "tensorflow/lite/micro/all_ops_resolver.h"
#include "tensorflow/lite/micro/micro_interpreter.h"
#include "tensorflow/lite/micro/system_setup.h"
#include "tensorflow/lite/schema/schema_generated.h"
#include "tensorflow/lite/micro/tflite_bridge/micro_error_reporter.h"

//network C files, only include one at a time
#include "exoLSTM4816_10q.h" //quantised simple LSTM
#include "exoMoveIdeal.h"

namespace {
    const tflite::Model* model = nullptr;
    tflite::MicroInterpreter* interpreter = nullptr;
    TfLiteTensor* model_input = nullptr;
    TfLiteTensor* model_output = nullptr;
    int32_t previous_time = 0;

    // Create an area of memory to use for input, output, and intermediate arrays.
    // The size of this will depend on the model you're using, and may need to be
    // determined by experimentation.
    constexpr int samples = 20; //number of samples per sensor
    constexpr int sensNum = 8; //number of sensors
    constexpr int features = (samples * sensNum); //total number of elements

    constexpr int kTensorArenaSize = 64 * 1024; //vary this with model size
    uint8_t tensor_arena[kTensorArenaSize];
    float feature_buffer[features]; //store features in buffer
    int8_t* model_input_buffer = nullptr;
[... 82 lines cut ...]
0x400da2f1: tflite::reference_ops::Concatenation (tflite::ConcatenationParams const&, tflite::RuntimeShape const* const*, signed char const* const*, tflite::RuntimeShape const&, signed char*) at Z:/My Files/PhDStuff/ESPArduino2/lib/tflite-lib/tensorflow/lite/kernels/internal/reference/concatenation.h line 43
0x400da4b7: tflite::(anonymous namespace)::EvalUnquantized (TfLiteContext*, TfLiteNode*) at Z:/My Files/PhDStuff/ESPArduino2/lib/tflite-lib/tensorflow/lite/micro/kernels/kernel_util.h line 64
0x400dab6a: tflite::(anonymous namespace)::Eval(TfLiteContext*, TfLiteNode*) at Z:/My Files/PhDStuff/ESPArduino2/lib/tflite-lib/tensorflow/lite/micro/kernels/concatenation.cc line 232
0x400fc61a: tflite::MicroGraph::InvokeSubgraph(int) at Z:/My Files/PhDStuff/ESPArduino2/lib/tflite-lib/tensorflow/lite/micro/micro_graph.cc line 174
0x400f706f: tflite::(anonymous namespace)::Eval(TfLiteContext*, TfLiteNode*) at Z:/My Files/PhDStuff/ESPArduino2/lib/tflite-lib/tensorflow/lite/micro/kernels/while.cc line 107
0x400fc61a: tflite::MicroGraph::InvokeSubgraph(int) at Z:/My Files/PhDStuff/ESPArduino2/lib/tflite-lib/tensorflow/lite/micro/micro_graph.cc line 174
0x400f7972: tflite::MicroInterpreter::Invoke() at Z:/My Files/PhDStuff/ESPArduino2/lib/tflite-lib/tensorflow/lite/micro/micro_interpreter.cc line 286
0x400d331a: setup() at Z:/My Files/PhDStuff/ESPArduino2/src/main.cpp line 88
0x400fee32: loopTask(void*) at C:/Users/tms40/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp line 42
```

Strangely, if I implement a single LSTM layer as opposed to two layers, such as the following:
```
model = tf.keras.Sequential([
    Input(shape=(20,8), name="Input"),
    LSTM(units=72, activation='relu',unroll=False),
    layers.Flatten(),
    Dense(units=pred, name="output")
    ])
```
Instead of getting the above error the program simply halts at _invoke_ and seemingly loops indefinitely (adding print functions to add.cc causes them to rapidly repeat)

Are there any current solutions to this, is this issue being worked on? Being able to reliably make LSTM Models for TFLite Micro would be useful for my current projects. If it is not currently feasible, are there any similar, small, portable alternatives?

Link to all code I use: [here](https://drive.google.com/drive/folders/1lExXH1mmNFC1bOwVJ564_nOGQN5YbApC?usp=sharing)

jmole, 2023-03-19:
Having this issue too, why doesn't ADD support int32?

jmole, 2023-03-19:
I managed to fix this in my local build, patch is attached.

Changes for the default **add.cc** kernel and the CMSIS-NN **add.cc** kernel are included here.

I haven't actually tried the default add.cc kernel because I am building with the CMSIS-NN library, but the implementation is effectively the same for both. This allowed me to run my LSTM tflite model on a cortex m7.

[0001-changes-to-add-to-support-int32-types-for-LSTM.patch](https://github.com/tensorflow/tflite-micro/files/11012191/0001-changes-to-add-to-support-int32-types-for-LSTM.patch)

ts532, 2023-03-19:
Hi there, This actually seems to work, thanks! I get neither crashes nor INT 32 Errors running this, instead actual predictions, although I have yet to test how good these predictions are. So far, I can only get the model to work if I make sure the entire model is converted to INT8 Format including inputs and outputs, which will cause issues with the normalised floats i'd prefer the input data format was, but that's an entirely different challenge to work on.

Hopefully this patch can be added to the main TF-Micro Repository.

Fabiran, 2023-03-22:
Is it the same solution for arduino nano 33 ble sense? I also meet this problem.

Fabiran, 2023-03-23:
> Is it the same solution for arduino nano 33 ble sense? I also meet this problem.

I modified the CMSIS-NN library\add.cpp, and the error didn't run ,but the output values is NaN. If I didn't modiflied the file, it would couse error: Type INT32 (2) not supported. Node ADD (number 0) failed to invoke with status 1. Node WHILE (number 10) failed to invoke with status 1. However, it had the correct output values.

closed 2023-04-23 by github-actions[bot]