tensorflow/tflite-micro #1885: When does the cmsis nn kernel of TFLM expand the support for other ops?

Raise-me-up, 2023-04-07:
As we all know, currently the cmsis nn kernel of TFLM only support the ops of add, conv, depthwise_conv, fully_connected, mul, pooling, softmax, svdf and unidirectional_sequence_lstm. When will other ops, such as transpose, transpose_conv, logistic, tanh, strided_slice, split, squared_difference, be supported?

mansnils, 2023-04-11:
The CMSIS-NN kernels consist of the most common kernels that make up the major parts of a typical model.
To illustrate the point it is not so useful to optimize a kernel that makes up 1 percent of a model.
Also some operators could not be much more optimized than equivalent ref kernel so might as well fall back on that.
That being said, once in a while new CMSIS-NN operators are added. LSTM is the last one that was added this year.
Is there any particular operator you would like have CMSIS-NN support for?

Raise-me-up, 2023-04-11:
@mansnils 
Thanks for your reply!  My model contains the ops of  transpose, transpose_conv and strided_slice, whick may lead to the slow speed of process. My target platform is cortex-m55, so the MVE technique is also desired. Is there any possibility for those ops to be supported in a short time?

mansnils, 2023-04-11:
Transpose_conv is a potential candidate for optimization but no concrete plans at the moment unfortunately. The others, transpose and strided_slice are memcpy operators so they will not be optimized.

Raise-me-up, 2023-04-12:
@mansnils 
Well, that sounds sadly. Maybe I shoud develop it myself. Is there any reference material recommended?

mansnils, 2023-04-12:
Contributions are welcome of course. There is not really any "reference material" but start with understanding conv and conv transpose and maybe take some inspiration from the current conv optimizations.

Raise-me-up, 2023-04-13:
@mansnils 
OK! Thanks for your detailed replys!

closed 2023-04-13 by Raise-me-up