tensorflow/tflite-micro #1943: How to accelerate the speed of LSTM op under the arm cortex-m55?

Raise-me-up, 2023-04-27:
Recently, I have found that the speed of LSTM op is a little slow under the arm cortex-m55, even though the MVE intrinsics have been used. Are there any other useful methods to accelerate the speed of it?

mansnils, 2023-05-02:
Hi @Raise-me-up 
I assume you use CMSIS-NN? Are you comparing to something else when you say a little slow?
Something that is faster and would accelerate it is https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/docs/arm.md#armr-ethostm-u-micronpu-family

Raise-me-up, 2023-05-06:
Hi @mansnils 
I do use CMSIS-NN to get a faster performance. However, the cost time of LSTM op is still unacceptable.

felix-johnny, 2023-05-08:
@Raise-me-up Thanks for the question and for using CMSIS-NN. I'd like to understand what you are observing to hopefully be able to suggest an improvement or explain what you see. 

1. What is the compiler(GCC/Armclang) and the version of it that you use?
2. What is the change in performance that you see? For e.g, if you consider the cycles for the entire network, what is (cycles_without_cmsis_nn / cycles_with_cmsis_nn). 
3. Is the activation type [int8](https://github.com/tensorflow/tflite-micro/blob/2e3be8531a40773c8190f93d87f96e271ff2dd10/tensorflow/lite/micro/kernels/cmsis_nn/unidirectional_sequence_lstm.cc#L596)? (int16 activation is not supported in CMSIS-NN yet)
4. To understand the level of vectorization, could you tell the values of n_input, n_output and n_batch as seen in https://github.com/tensorflow/tflite-micro/blob/2e3be8531a40773c8190f93d87f96e271ff2dd10/tensorflow/lite/micro/kernels/cmsis_nn/unidirectional_sequence_lstm.cc#L461
5. What is the compiler optimization level that you set?

Raise-me-up, 2023-05-09:
Hi @felix-johnny! Thanks for your reply! My answers are as follows:

1.  I use arm-none-eabi-gcc and arm-none-eabi-g++  to build the codes, whose version is 
(GNU Arm Embedded Toolchain 10-2020-q4-major) 10.2.1 20201103 (release)
2. Sorry, I don't understand the meaning of the cycles for the entire network here. The performance I refer to means the process time of LSTM layer.
3. Of course, the activation type is int8.
4. I check the results as follows:
(1) LSTM1:  
n_input: 16     n_output: 16    n_batch: 1   process time:  cost 659us
(2) LSTM2:
n_input: 16    n_output: 16    n_batch: 16    process time:  cost 663us
5.  The optimization level I set is O2 level

felix-johnny, 2023-05-11:
@Raise-me-up 
for point 2, it is ok that you use time taken for a layer instead of the whole network. What is the layer processing time without CMSIS-NN? 

As for the performance that you get, there are a couple of reasons and options to improve it.
1. 
For Arm Cortex-M55 there is a difference in the compilers vectorization capability between armclang and GCC(atleast till version 12.x). For example in the case of LSTM, a function or segment of code that does a good part of the processing is https://github.com/ARM-software/CMSIS-NN/blob/main/Source/NNSupportFunctions/arm_nn_vec_mat_mul_result_acc_s8.c#L79
A comparison of the disassembly for the inner loop generated generated by Armclang and GCC along with the command used is shown below. 

[image]

 The efficient vectorization by Armclang gives a better performance here. That is one option to improve the performance. But, I believe the performance for GCC(Cortex-M55 context) should see a boost with the upcoming GCC 13 release. 

2.The n_input and n_output of 16 translates to the inner loop in line 79(same link as above) being executed once. Considering the setting up of the loop for one iteration, that is something that can improved if you are after relative uplift.

closed 2023-06-11 by github-actions[bot]