tensorflow/tflite-micro #1973: import error tflite_micro

aryanxk02, 2023-05-16:
I am trying to run `generate_micro_mutable_op_resolver_from_model.py` file via bazel. But seems like I am getting an import error.

```
ModuleNotFoundError: No module named 'tflite_micro'
```

Reference: https://github.com/aryanxk02/tflite-micro/blob/main/tensorflow/lite/micro/tools/gen_micro_mutable_op_resolver/generate_micro_mutable_op_resolver_from_model.py#L26

aryanxk02, 2023-05-16:
Okay some tweaks here. I changed the import statement from
```
from tflite_micro.tensorflow.lite.tools import visualize as visualize
```
to
```
from tensorflow.lite.tools import visualize as visualize
```

It works now.

vikramdattu, 2023-05-22:
cc @advaitjain I think this is a bug and import should have been `tensorflow.lite.tools` and not `tflite_micro.`

closed 2024-04-18 by mansnils