tensorflow/tflite-micro #1979: Bazel not build - "flatbuffers/flatbuffers.h" no such file or directory

Gabrielhcouto, 2023-05-17:
Hello!

I'm trying to run the Bazel build scripts, but I've never used Bazel before, so I'm not sure if I'm doing something wrong. Explaining better, I'm want to run a CNN on an ESP32-S3, but when compiling the code using the `tensorflow` and `third_party` folders I get an include error:

```` c
In file included from lib/tflitemicro/tensorflow/lite/core/api/flatbuffer_conversions.h:28,
                 from lib/tflitemicro/tensorflow/lite/core/api/flatbuffer_conversions.cc:16:
lib/tflitemicro/tensorflow/lite/schema/schema_generated.h:7:10: fatal error: flatbuffers/flatbuffers.h: No such file or directory
 #include "flatbuffers/flatbuffers.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\esp32s3box\lib702\tflitemicro\tensorflow\lite\core\api\flatbuffer_conversions.cc.o] Error 1
In file included from lib/tflitemicro/tensorflow/lite/experimental/microfrontend/lib/fft.cc:19:
lib/tflitemicro/tensorflow/lite/experimental/microfrontend/lib/kiss_fft_int16.h:26:10: fatal error: kiss_fft.h: No such file or directory
````
in the `flatbuffer_conversions.cc` code on line 22, the `flatfbuffers.h` library is included and there is a comment with `from @flatbuffers`, so I imagine that the bazel builds did not generate the necessary files since the `@flatbuffers` file exists on my computer at: `C:\Users\developer\_bazel_developer\27uxfnsm\external\@flatbuffers.marker` <br/>

I cloned this repository, downloaded Bazel on my computer and using a `git bash` I navigated to the root of the repository and executed the command:

````
touch WORKSPACE
````

this doesn't return anything, it just executes. Then I ran the next command:
````
bazel build //...
````
this is where I get the error:
```` bash
$ bazel build //...
Starting local Bazel server and connecting to it...
Loading:
Loading:
Loading:
Loading: 0 packages loaded
Analyzing: 1159 targets (56 packages loaded)
Analyzing: 1159 targets (56 packages loaded, 0 targets configured)
Analyzing: 1159 targets (64 packages loaded, 462 targets configured)
ERROR: C:/users/developer/_bazel_developer/27uxfnsm/external/tflm_pip_deps_tensorflow_cpu/BUILD.bazel:22:11: no such package '@tflm_pip_deps_tensorflow_intel//': The repository '@tflm_pip_deps_tensorflow_intel' could not be resolved: Repository '@tflm_pip_deps_tensorflow_intel' is not defined and referenced by '@tflm_pip_deps_tensorflow_cpu//:pkg'
ERROR: Analysis of target '//tensorflow/lite/tools:test_utils' failed; build aborted:
INFO: Elapsed time: 11.759s
INFO: 0 processes.
FAILED: Build did NOT complete successfully (97 packages loaded, 852 targets configured)
````
I didn't understand very well, but it seems to me that some build is not finding the file `@tflm_pip_deps_tensorflow_intel`, so I looked inside all the folders and really that file doesn't exist and I didn't find it here in the repository either. <br/>

but I'm not sure if the error is because of bazel or if the `flatbuffers/flatbuffers.h` file should exist. **My doubt is do I really need to build these files to be able to compile the tfmicro library?** <br/>

I'm using `Visual Studio Code` with `PlatformIO` to compile C code. I just copied the tflite-micro folder and put it in the project's lib folder.

Gabrielhcouto, 2023-05-17:
I believe I solved my problem, it was really inclusion. But I used the repository folders [tflite-micro-esp-examples](https://github.com/espressif/tflite-micro-esp-examples), In this repository there are some examples of use. Even using this repository I had to modify some inclusion directories so that my code in C finds the header files.

I was able to compile and build the project within ESP32-S3. I had to make some changes to my code to adjust the new parameters, because I was using a very old Library of tflite-micro. <br/>

When running my code I am receiving another error using the "resolves":
````
Didn't find op for builtin opcode 'RESHAPE'
Failed to get registration from op code SHAPE
````
But I'm including 
````
resolver = new tflite::MicroMutableOpResolver<6>();
    resolver->AddFullyConnected();
    resolver->AddMaxPool2D();
    resolver->AddConv2D();
    resolver->AddSoftmax();
    resolver->AddRelu();
    resolver->AddReshape();
````
I believe I can find in another solution elsewhere, or this can be a subject for a new issue.

closed 2023-06-18 by github-actions[bot]

jjgescalante, 2023-07-28:
Hello, i guess you need to add shape opcode
resolver->addShape()

In this version for tensorflow micro opcode shape is added.

I'm still fighting the same battle for new tensor flow models.