tensorflow/tflite-micro #2005: Project failed to build because messing file "fixedpoint/fixedpoint.h"

fathiabdelmalek, 2023-05-28:
I want to use tflite-micro library on an Arduino project using PlatformIO extension on VS Code, I have cloned the library from github and added it to `lib` folder, here is my code:
```
#include "tensorflow/lite/micro/micro_mutable_op_resolver.h"
#include "tensorflow/lite/micro/micro_interpreter.h"
#include "tensorflow/lite/schema/schema_generated.h"
```
but when I run build, I get this error:
```
lib/tflite-micro/tensorflow/lite/kernels/internal/common.h:28:10: fatal error: fixedpoint/fixedpoint.h: No such file or directory
```
I'm new to tflite-micro, but it seems that the problem is that there is a missed library or file, please help

mbernico, 2023-05-28:
Hi there.  This is very likely due to a path problem caused by putting tflm in the lib folder.   You might need to adjust your build system...

It might be easier to use the supported Arduino library?

https://github.com/tensorflow/tflite-micro-arduino-examples

Otherwise check out the new platforms guide:
https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/docs/new_platform_support.md

fathiabdelmalek, 2023-05-28:
@mbernico thanks for the reply, I have fixed the problem by manually adding the messing libraries. Now I have another question is just cloning the GitHub repository on the lib folder enough to use it or should I build it another way?

closed 2023-05-29 by fathiabdelmalek