tensorflow/tflite-micro #2058: c++17-only *inline variable* feature introduced in sync from upstream TF

rkuester, 2023-06-15:
The sync from upstream TF in PR #2053 brought the c++17-only feature *inline variables* into code widely included in the TFLM kernels: 

https://github.com/tensorflow/tflite-micro/blob/857842e6b184819011cb2cfe59899b7b97e22785/tensorflow/lite/core/macros.h#L68-L76

This causes the following warning in my bazel builds. Using inline variables might lead to build failures with other compilers, or at worst: unexpected behavior if the compiler ignores the intent of `inline` with or without warning. (Although in this case, the variables are const, so maybe it'll just waste space :shrug:.) . BTW, I'm working on turning on warnings into errors a la `-Werror` in #2057 so that a warning like this, subtly introduced in a sync from upstream TF, won't pass CI.

````
ERROR: /home/rkuester/wetu/tflite-micro/tensorflow/lite/micro/kernels/BUILD:146:11: Compiling tensorflow/lite/micro/kernels/micro_tenso
r_utils.cc failed: (Exit 1): gcc failed: error executing command (from target //tensorflow/lite/micro/kernels:micro_tensor_utils) /usr/
bin/gcc -U_FORTIFY_SOURCE -fstack-protector -Wall -Wunused-but-set-parameter -Wno-free-nonheap-object -fno-omit-frame-pointer '-std=c++
0x' -MD -MF ... (remaining 22 arguments skipped)

Use --sandbox_debug to see verbose messages from the sandbox and retain the sandbox build root for debugging
In file included from ./tensorflow/lite/kernels/internal/common.h:29,
                 from tensorflow/lite/micro/kernels/micro_tensor_utils.cc:25:
./tensorflow/lite/core/macros.h:70:1: error: inline variables are only available with '-std=c++17' or '-std=gnu++17' [-Werror=c++17-ext
ensions]
   70 | inline constexpr unsigned int flatbuffer_size_max = 2147483648;
      | ^~~~~~
./tensorflow/lite/core/macros.h:72:1: error: inline variables are only available with '-std=c++17' or '-std=gnu++17' [-Werror=c++17-ext
ensions]
   72 | inline constexpr char tflite_metadata_buffer_location[] = "buffer_location";
      | ^~~~~~
./tensorflow/lite/core/macros.h:74:1: error: inline variables are only available with '-std=c++17' or '-std=gnu++17' [-Werror=c++17-ext
ensions]
   74 | inline constexpr char tflite_metadata_min_runtime_version[] =
      | ^~~~~~
cc1plus: all warnings being treated as errors
````

rkuester, 2023-06-15:
Pinging @advaitjain, @rascani.

advaitjain, 2023-06-15:
Right, there should be a `copts = micro_copts()` in this BUILD rule https://github.com/tensorflow/tflite-micro/blob/857842e6b184819011cb2cfe59899b7b97e22785/tensorflow/lite/micro/kernels/BUILD#L146-L159

advaitjain, 2023-06-15:
alternately, perhaps a better solution is to add -std=c++11 to https://github.com/tensorflow/tflite-micro/blob/main/.bazelrc?

rkuester, 2023-06-15:
> alternately, perhaps a better solution is to add -std=c++11

You mean c++17?

Makes sense to me if we really intend for **all** bazel builds to now be c++17. Is that change in policy explained somewhere?

rkuester, 2023-06-15:
> alternately, perhaps a better solution is to add -std=c++11 to [.bazelrc]

Because from my testing this applies to many more targets than just `//tensorflow/lite/micro/kernels:micro_tensor_utils`.

PR #2060 (2023-06-15): build(bazel): use --std=c++17 by default
Make all bazel builds use --std=c++17 by default.

BUG=fixes #2058
commit 52c95681 (2023-06-15): build(bazel): use --std=c++17 by default (#2060)

Make all bazel builds use --std=c++17 by default.

BUG=fixes #2058