tensorflow/tflite-micro #2079: Requested feature_data_ size 536907080 doesn't match 1960; Feature generation failed;

Svobi27, 2023-06-23:
Hello Together,

I'm having a Problem with the micro_speech example for arduino from this repo: https://github.com/tensorflow/tflite-micro-arduino-examples/tree/main/examples/micro_speech

When trying to use this example with a new trained model from this jupyter noteboobk: https://github.com/tensorflow/tflite-micro/tree/main/tensorflow/lite/micro/examples/micro_speech/train

I always get the same error message: 
Requested feature_data_ size 536907080 doesn't match 1960
Feature generation failed

The only thing i changed in the notebook was the tensorflow version. This is because this notebook was using 1.x Version which is no longer supported by colab and i changed it to work with the latest 2.x version

Can anyone help here?

Greetings,
Patrick

Svobi27, 2023-08-08:
Any Update on this Bug?

closed 2023-09-08 by github-actions[bot]

avisinghal6, 2023-11-24:
Is this issue solved?

Lieber0402, 2024-01-22:
I've found something that may cause this error, in the original file model.cc in the version r2.6 tensorflow, the size of the g_model_len is 18712, and the size in the trained model and the model created by the checkpoint file is 18960. This may caused this problem, but I still don't understand how to fix it. And the tensorflow version I use to train on the colab is 2.15.0 btw.

Lieber0402, 2024-01-23:
> I've found something that may cause this error, in the original file model.cc in the version r2.6 tensorflow, the size of the g_model_len is 18712, and the size in the trained model and the model created by the checkpoint file is 18960. This may caused this problem, but I still don't understand how to fix it. And the tensorflow version I used to train on the colab is 2.15.0 btw.

https://github.com/tensorflow/tensorflow/issues/44868#issuecomment-1905153457 The model deployment error has been solved, but the Sparkfun Edge board can't recognize the keywords "yes" and "no" so well.