tensorflow/tflite-micro #2099: [Micro Speech Example] tensorflow/lite/micro/tools/make/targets/disco_f746ng_makefile.inc: No such file or directory

aravind-pp, 2023-06-30:
Hi,

I am trying to build [Micro Speech Example](https://github.com/tensorflow/tflite-micro/tree/main/tensorflow/lite/micro/examples/micro_speech) for STM32F7 discovery kit. I installed Mbed CLI on Ubuntu 22.04.2 LTS and tried creating folder structure using this command:
`make -f tensorflow/lite/micro/tools/make/Makefile TARGET=disco_f746ng OPTIMIZED_KERNEL_DIR=cmsis_nn generate_micro_speech_mbed_project`

I am getting this build error:
```
tensorflow/lite/micro/tools/make/Makefile:555: tensorflow/lite/micro/tools/make/targets/disco_f746ng_makefile.inc: No such file or directory
make: *** No rule to make target 'tensorflow/lite/micro/tools/make/targets/disco_f746ng_makefile.inc'.  Stop.
```

Please help.

Thank you,
Aravind.

aravind-pp, 2023-07-26:
Hi,

This is a followup. Could anyone help in resolving above issue?

Thank you,
Aravind.

mansnils, 2023-08-07:
Hi @aravind-pp 
It looks like that documentation is a bit outdated. Here are the official supported platforms: https://github.com/tensorflow/tflite-micro#official-builds
For other targets than that here is a guideline: https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/docs/new_platform_support.md

aravind-pp, 2023-08-08:
Hi @mansnils ,

Thank you for providing above links. I will take a look at them.

closed 2023-09-07 by github-actions[bot]