tensorflow/tflite-micro #2101: VAR_HANDLE requires resource variables. Please create ResourceVariables and pass it to the interpreter.

Raise-me-up, 2023-07-05:
As shown in the question, how to solve the problem? The problem occurs when the stateful of LSTM is set to true.

yair-ehrenwald, 2023-07-05:
Haven't tried it with LSTM yet, but something like this works for me in other cases:

uint8_t var_arena[4096];
tflite::MicroAllocator *ma = tflite::MicroAllocator::Create(var_arena, 4096);
tflite::MicroResourceVariables *mrv = tflite::MicroResourceVariables::Create(ma, 20);

interpreter = new tflite::MicroInterpreter(model, micro_op_resolver, tensor_arena, kTensorArenaSize, mrv );

Raise-me-up, 2023-07-06:
@yair-ehrenwald It also works for me, thank you very much!

closed 2023-07-06 by Raise-me-up

ctwillson, 2024-10-15:
@Raise-me-up Have u ever quant the stateful model?