tensorflow/tflite-micro #2120: `MicroProfiler` not timing operations correctly in the Coral Dev Board Micro

fabito, 2023-07-13:
### Description

I am trying to profile my model operations running on the coralmicro dev board.
I am injecting a [`tflite::MicroProfiler`](https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/micro_profiler.h) into the `tflite::MicroInterpreter`:

```cpp
  tflite::MicroProfiler profiler;
  tflite::MicroErrorReporter error_reporter;
  tflite::MicroMutableOpResolver<3> resolver;
  resolver.AddDequantize();
  resolver.AddDetectionPostprocess();
  resolver.AddCustom(kCustomOp, RegisterCustomOp());

  tflite::MicroInterpreter interpreter(tflite::GetModel(model.data()), resolver,
                                       tensor_arena, kTensorArenaSize,
                                       &error_reporter, nullptr,
                                       &profiler // injecting profiler via constructor
);
```
Then I run inference and print the profiling results:

```cpp
std::memcpy(tflite::GetTensorData<uint8_t>(input_tensor), image->data(), image->size());
if (interpreter->Invoke() != kTfLiteOk) return false;
*results = tensorflow::GetDetectionResults(interpreter, 0.5, 1);

// logging profile results
profiler.Log();
```

See below the profiler output. Notice that latency is always `0 ms`.

```
edgetpu-custom-op took 0 ticks (0 ms).
DEQUANTIZE took 0 ticks (0 ms).
DEQUANTIZE took 0 ticks (0 ms).
TFLite_Detection_PostProcess took 0 ticks (0 ms).
```

Is [`micro_time.h`](https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/micro_time.h) properly implemented for the dev board micro ?
Is there anything else I need to do ( maybe a specific compile flag ? ) to enable profiling ?

yair-ehrenwald, 2023-07-13:
I'm not familiar with the corel dev board, but take a look in the vendor specific directories under 
https://github.com/tensorflow/tflite-micro/tree/main/tensorflow/lite/micro
that should give you an idea of what you need to implement for your board:

uint32_t ticks_per_second() 
uint32_t GetCurrentTimeTicks()

fabito, 2023-07-16:
Thanks, I am targetting the cortex-m7, I will try `cortex_m_generic/micro_time.cc`

fabito, 2023-07-16:
After referencing [cortex_m_generic/micro_time.cc](https://github.com/tensorflow/tflite-micro/blob/52007f6a479a37c6513fa5b6523791572902c25c/tensorflow/lite/micro/cortex_m_generic/micro_time.cc#L38) I started getting ticks:

```
edgetpu-custom-op took 75724427 ticks (-1 ms).
DEQUANTIZE took 297247 ticks (-1 ms).
DEQUANTIZE took 6639784 ticks (-1 ms).
```

millisecond convertion still broken though, apparently `ticks_per_second` [always return 0](https://github.com/tensorflow/tflite-micro/blob/52007f6a479a37c6513fa5b6523791572902c25c/tensorflow/lite/micro/micro_time.h#L29).

closed 2023-07-16 by fabito