tensorflow/tflite-micro #2126: Error when compile with arm-none-eabi-gcc-9.2.1

hmVictorYu, 2023-07-14:
Is it possible to build tflite micro with arm-none-eabi-gcc-9.2.1, when i build the library use the defualt download gcc ererything works fine. Since the whole project of mine, which i want to add tflm in is build use arm-none-eabi-gcc-9.2.1, so i change the gcc version to arm-none-eabi-gcc-9.2.1 in tensorflow/lite/micro/tools/make/downloads/gcc_embedded/  that cause the build error.
Here is the make command i use:

`make -f tensorflow/lite/micro/tools/make/Makefile BUILD_TYPE=release TARGET=cortex_m_generic TARGET_ARCH=cortex-m33 OPTIMIZED_KERNEL_DIR=cmsis_nn microlite -j8
`

and the errors occurs below
`In file included from tensorflow/lite/micro/tools/make/downloads/cmsis_nn/Include/arm_nnsupportfunctions.h:33,
                 from tensorflow/lite/micro/tools/make/downloads/cmsis_nn/Source/ActivationFunctions/arm_relu_q15.c:32:
tensorflow/lite/micro/tools/make/downloads/cmsis_nn/Source/ActivationFunctions/arm_relu_q15.c: In function 'arm_relu_q15':
tensorflow/lite/micro/tools/make/downloads/cmsis_nn/Include/Internal/arm_nn_compiler.h:187:20: error: implicit declaration of function '__qsub16' [-Werror=implicit-function-declaration]
  187 |     #define QSUB16 __qsub16
      |                    ^~~~~~~~
tensorflow/lite/micro/tools/make/downloads/cmsis_nn/Source/ActivationFunctions/arm_relu_q15.c:71:16: note: in expansion of macro 'QSUB16'
   71 |         mask = QSUB16(0x00000000, buf);
      |                ^~~~~~
tensorflow/lite/micro/tools/make/downloads/cmsis_nn/Source/ActivationFunctions/arm_relu_q15.c: At top level:
cc1: error: unrecognized command line option '-Wno-unused-private-field' [-Werror]
cc1: all warnings being treated as errors
tensorflow/lite/micro/tools/make/Makefile:709: recipe for target 'gen/cortex_m_generic_cortex-m33_release/obj/third_party_kernels/tensorflow/lite/micro/tools/make/downloads/cmsis_nn/Source/ActivationFunctions/arm_relu_q15.o' failed
make: *** [gen/cortex_m_generic_cortex-m33_release/obj/third_party_kernels/tensorflow/lite/micro/tools/make/downloads/cmsis_nn/Source/ActivationFunctions/arm_relu_q15.o] Error 1
make: *** Waiting for unfinished jobs....
`

any ideal to solve it?

hmVictorYu, 2023-07-14:
also happens to other kernel in cmsis-nn
`
                 from tensorflow/lite/micro/tools/make/downloads/cmsis_nn/Source/ActivationFunctions/arm_relu_q15.c:32:
tensorflow/lite/micro/tools/make/downloads/cmsis_nn/Source/ActivationFunctions/arm_relu_q15.c: In function 'arm_relu_q15':
tensorflow/lite/micro/tools/make/downloads/cmsis_nn/Include/Internal/arm_nn_compiler.h:187:20: error: implicit declaration of function '__qsub16' [-Werror=implicit-function-declaration]
  187 |     #define QSUB16 __qsub16
      |                    ^~~~~~~~
tensorflow/lite/micro/tools/make/downloads/cmsis_nn/Source/ActivationFunctions/arm_relu_q15.c:71:16: note: in expansion of macro 'QSUB16'
   71 |         mask = QSUB16(0x00000000, buf);
      |                ^~~~~~
In file included from tensorflow/lite/micro/tools/make/downloads/cmsis_nn/Include/arm_nnsupportfunctions.h:33,
                 from tensorflow/lite/micro/tools/make/downloads/cmsis_nn/Source/ActivationFunctions/arm_relu_q7.c:32:
tensorflow/lite/micro/tools/make/downloads/cmsis_nn/Source/ActivationFunctions/arm_relu_q7.c: In function 'arm_relu_q7':
tensorflow/lite/micro/tools/make/downloads/cmsis_nn/Include/Internal/arm_nn_compiler.h:186:19: error: implicit declaration of function '__qsub8' [-Werror=implicit-function-declaration]
  186 |     #define QSUB8 __qsub8
      |                   ^~~~~~~
tensorflow/lite/micro/tools/make/downloads/cmsis_nn/Source/ActivationFunctions/arm_relu_q7.c:71:16: note: in expansion of macro 'QSUB8'
   71 |         mask = QSUB8(0x00000000, buf);
      |                ^~~~~
In file included from tensorflow/lite/micro/tools/make/downloads/cmsis_nn/Include/arm_nnsupportfunctions.h:33,
                 from tensorflow/lite/micro/tools/make/downloads/cmsis_nn/Source/BasicMathFunctions/arm_elementwise_mul_s8.c:32:
tensorflow/lite/micro/tools/make/downloads/cmsis_nn/Source/BasicMathFunctions/arm_elementwise_mul_s8.c: In function 'arm_elementwise_mul_s8':
tensorflow/lite/micro/tools/make/downloads/cmsis_nn/Include/Internal/arm_nn_compiler.h:188:20: error: implicit declaration of function '__sadd16' [-Werror=implicit-function-declaration]
  188 |     #define SADD16 __sadd16
      |                    ^~~~~~~~
tensorflow/lite/micro/tools/make/downloads/cmsis_nn/Source/BasicMathFunctions/arm_elementwise_mul_s8.c:119:15: note: in expansion of macro 'SADD16'
  119 |         a_1 = SADD16(a_1, offset_1_packed);
      |               ^~~~~~
tensorflow/lite/micro/tools/make/downloads/cmsis_nn/Source/ActivationFunctions/arm_relu_q15.c: At top level:
cc1: error: unrecognized command line option '-Wno-unused-private-field' [-Werror]
cc1: all warnings being treated as errors
tensorflow/lite/micro/tools/make/Makefile:709: recipe for target 'gen/cortex_m_generic_cortex-m33_release/obj/third_party_kernels/tensorflow/lite/micro/tools/make/downloads/cmsis_nn/Source/ActivationFunctions/arm_relu_q15.o' failed
make: *** [gen/cortex_m_generic_cortex-m33_release/obj/third_party_kernels/tensorflow/lite/micro/tools/make/downloads/cmsis_nn/Source/ActivationFunctions/arm_relu_q15.o] Error 1
make: *** Waiting for unfinished jobs....
tensorflow/lite/micro/tools/make/downloads/cmsis_nn/Source/ActivationFunctions/arm_relu_q7.c: At top level:
cc1: error: unrecognized command line option '-Wno-unused-private-field' [-Werror]
cc1: all warnings being treated as errors
tensorflow/lite/micro/tools/make/Makefile:709: recipe for target 'gen/cortex_m_generic_cortex-m33_release/obj/third_party_kernels/tensorflow/lite/micro/tools/make/downloads/cmsis_nn/Source/ActivationFunctions/arm_relu_q7.o' failed
make: *** [gen/cortex_m_generic_cortex-m33_release/obj/third_party_kernels/tensorflow/lite/micro/tools/make/downloads/cmsis_nn/Source/ActivationFunctions/arm_relu_q7.o] Error 1
tensorflow/lite/micro/tools/make/downloads/cmsis_nn/Source/BasicMathFunctions/arm_elementwise_mul_s8.c: At top level:
cc1: error: unrecognized command line option '-Wno-unused-private-field' [-Werror]
cc1: all warnings being treated as errors
tensorflow/lite/micro/tools/make/Makefile:709: recipe for target 'gen/cortex_m_generic_cortex-m33_release/obj/third_party_kernels/tensorflow/lite/micro/tools/make/downloads/cmsis_nn/Source/BasicMathFunctions/arm_elementwise_mul_s8.o' failed
make: *** [gen/cortex_m_generic_cortex-m33_release/obj/third_party_kernels/tensorflow/lite/micro/tools/make/downloads/cmsis_nn/Source/BasicMathFunctions/arm_elementwise_mul_s8.o] Error 1
`

mansnils, 2023-08-09:
Hi @hmVictorYu 
What host platform are you building on, e.g. Linux and what is the download link?
If you could provide a draft PR or copy paste the code where the downloaded version is updated here, it would help: https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/tools/make/arm_gcc_download.sh

closed 2023-09-10 by github-actions[bot]