tensorflow/tflite-micro #2133: Converting *.tflite to *.cc for Arduino Nano 33 Sense

matz01952, 2023-07-18:
Python version 3.10.12
TensorFlow version 2.12.0

I created a Keras model using the [transfer learning tutorial](https://www.tensorflow.org/tutorials/images/transfer_learning). I then used the quantization and converter from the [TinyML Book](https://tinymlbook.com/) to produce an array of bytes for my Arduino nano sense with the ArduCam 2MP Plus camera.

Using [this](https://projecthub.arduino.cc/anaferraz/embedded-machine-learning-for-person-detection-ee255f) Arduino code example to test my board and camera, which worked. I then modified the byte array with my own. This gave me the error “Invoke() called after initialization failed”. Further investigation printing out “TF_LITE_REPORT_ERROR(error_reporter, “Bad input tensor, size. %d”, input->dims->size);” produced the value 2113404085. The same line with the original byte array produced 4.

When I pull the latest tflite-micro repo and use the person detection example the tensor size is still the same 2113404085 value.

Putting my tflite file into Netron showed my input to be (1, 224, 224, 3) shape with float32 data type.

Therefore I presume that my error is in the byte array. Can anyone shed some light on this, please?

Some example code I have made attempts with:
```
model = tf.keras.models.load_model(modelDir_fine)
concrete_func = model.signatures[tf.saved_model.DEFAULT_SERVING_SIGNATURE_DEF_KEY]
concrete_func.inputs[0].set_shape([1, 224, 224, 3])
converter = tf.lite.TFLiteConverter.from_concrete_functions([concrete_func])
converter.optimizations = [tf.lite.Optimize.DEFAULT]
tfl_model = converter.convert()

open("model.tflite", "wb").write(tfl_model)

!apt-get update && apt-get -qq install xxd
!xxd -i model.tflite > model.cc
```

```
model = tf.keras.models.load_model(modelDir_fine)
converter = tf.lite.TFLiteConverter.from_keras_model(model)
converter.optimizations = [tf.lite.Optimize.DEFAULT]
tfl_model = converter.convert()

open("model.tflite", "wb").write(tfl_model)

!apt-get update && apt-get -qq install xxd
!xxd -i model.tflite > model.cc
```

```
  repr_ds = ds_rep.unbatch()
  def representative_data_gen():
    for i_value, o_value in repr_ds.batch(1):
      yield [i_value]
  converter = tf.lite.TFLiteConverter.from_saved_model(modelDir_fine)
  converter.optimizations = [tf.lite.Optimize.DEFAULT]
  converter.target_spec.supported_ops = [tf.lite.OpsSet.TFLITE_BUILTINS_INT8]
  converter.inference_input_type = tf.int8
  converter.inference_output_type = tf.int8
  converter.representative_dataset = tf.lite.RepresentativeDataset(representative_data_gen)
  tfl_model = converter.convert()

  open("model.tflite", "wb").write(tfl_model)

!apt-get update && apt-get -qq install xxd
!xxd -i model.tflite > model.cc
```

This is my model [CheckPoint-16-0.0282.hdf5.zip](https://github.com/tensorflow/tflite-micro/files/12088877/CheckPoint-16-0.0282.hdf5.zip)

adityarajsahu, 2023-08-07:
Hi @matz01952 , how are you getting the inference from the model.cc ?

matz01952, 2023-08-07:
Hi @adityarajsahu,

I have declared the model variable like this:
`const tflite::Model* model = nullptr;`
I am getting the model using this function:
`model = tflite::GetModel(g_person_detect_model_data);`
Where `g_person_detect_model_data` is my byte array.
I have then built the interpreter with this function:
`static tflite::MicroInterpreter static_interpreter(
      model, resolver, tensor_arena, kTensorArenaSize);`
Then set the interpreter:
  `interpreter = &static_interpreter;`
To get the input I have set a TfLiteTensor like this:
`input = interpreter->input(0);`
To get the data from the camera I have called:
`GetImage(error_reporter, kNumCols, kNumRows, kNumChannels, input->data.int8)`.

Thanks

Matt

closed 2023-09-07 by github-actions[bot]