tensorflow/tflite-micro #2164: Problem with Makefile generate_projects

macso-vincent-russell, 2023-08-11:
Following:

git clone https://github.com/tensorflow/tflite-micro.git
cc tflite-micro
make -f tensorflow/lite/micro/tools/make/Makefile generate_projects

I'm running into:
tensorflow/lite/micro/tools/make/downloads/flatbuffers already exists, skipping the download.
tensorflow/lite/micro/tools/make/downloads/kissfft already exists, skipping the download.
tensorflow/lite/micro/tools/make/downloads/pigweed already exists, skipping the download.
make: *** No rule to make target 'generate_projects'.  Stop.

This error happened on WSL2 and a Ubuntu Virtual Machine.

Velocities, 2023-08-12:
I'm having this issue too on my Ubuntu Virtual Machine, I hope this gets resolved soon.

misterb0407, 2023-08-14:
Same issue for me, using WSL2 on Win 11

basahiy, 2023-08-15:
> Trying to create a new project following the instructions in https://www.tensorflow.org/lite/microcontrollers/library
> 
> Doing these commands:
> 
> git clone https://github.com/tensorflow/tflite-micro.git cc tflite-micro make -f tensorflow/lite/micro/tools/make/Makefile generate_projects
> 
> I'm running into: tensorflow/lite/micro/tools/make/downloads/flatbuffers already exists, skipping the download. tensorflow/lite/micro/tools/make/downloads/kissfft already exists, skipping the download. tensorflow/lite/micro/tools/make/downloads/pigweed already exists, skipping the download. make: *** No rule to make target 'generate_projects'. Stop.
> 
> This error happened on two separate computers on WSL2 and a Ubuntu Virtual Machine, all in Ubuntu 20.04.6 LTS.

please replace “generate_projects” with the project name.(for example: test_hello_world_test)

atakli, 2023-08-27:
@basahiy  So? Where is the hello world project generated?

closed 2023-09-27 by github-actions[bot]