tensorflow/tflite-micro #2196: AllocateTensors fails with "Didn't find op for builtin opcode 'QUANTIZE'" despite resolver.AddQuantize()

lioff, 2023-08-29:
Hi all,

I updated tflite-micro and switched from AllOpsResolver to MicroMutableOpResolver.

However, this always results in the following error:

```
Didn't find op for builtin opcode 'QUANTIZE'
Failed to get registration from op code QUANTIZE
```

Either I got something wrong or the support for quantization aware models is broken?

I also tried to retrain and convert my model with the most recent release of tensorflow_model_optimization.

Kind regards,
Lion

See my code below (using ESP32-S3):

```tflite::MicroMutableOpResolver<5> resolver;
if (
    resolver.AddFullyConnected() != kTfLiteOk &&
    resolver.AddConv2D() != kTfLiteOk &&
    resolver.AddQuantize() != kTfLiteOk &&
    resolver.AddDequantize() != kTfLiteOk &&
    resolver.AddRelu() != kTfLiteOk
) {
    ESP_LOGE(TAG, "error in setup of resolver\n");
}

const tflite::Model* model = ::tflite::GetModel(model_tflite);
if (model->version() != TFLITE_SCHEMA_VERSION) {
    ESP_LOGE(TAG, "wrong TFLITE_SCHEMA_VERSION\n");
}

uint8_t *tensor_arena = (uint8_t *) heap_caps_malloc(kTensorArenaSize, MALLOC_CAP_8BIT | MALLOC_CAP_SPIRAM);
if (tensor_arena == NULL) {
    ESP_LOGE(TAG, "couldn't allocate memory of %d bytes\n", kTensorArenaSize);
}

tflite::MicroInterpreter interpreter(model, resolver, tensor_arena, kTensorArenaSize);

if (interpreter.AllocateTensors() != kTfLiteOk) {
    ESP_LOGE(TAG, "error while allocating tensors\n");
}```

oniigiirii, 2023-08-29:
Hello

I am currently facing the same issue but with a 2D Convolutional layer instead.

This is the error I get:
```
Didn't find op for builtin opcode 'CONV_2D'
Failed to get registration from op code CONV_2D
```

and this is my code for registering the operations:

```
        ...
        auto statusConv2d = resolver.AddConv2D();
        auto statusDepthwiseConv2d = resolver.AddDepthwiseConv2D();
        auto statusRelu = resolver.AddRelu();
        auto statusGlobalAverage = resolver.AddAveragePool2D();
        auto statusFC = resolver.AddFullyConnected();

        /// Verify Resolver Operation addition
        bool setUpSuccessful = true;

        if (statusConv2d != kTfLiteOk)
        {
            dsp_print_error("Failed while adding Conv2d Layer to OperationResolver.");
            setUpSuccessful = false;
        }

        if (statusDepthwiseConv2d != kTfLiteOk)
        {
            dsp_print_error("Failed while adding DepthwiseConv2d Layer to OperationResolver.");
            setUpSuccessful = false;
        }

        if (statusRelu != kTfLiteOk)
        {
            dsp_print_error("Failed while adding ReLU Activation Function to OperationResolver.");
            setUpSuccessful = false;
        }

        if (statusGlobalAverage != kTfLiteOk)
        {
            dsp_print_error("Failed while adding Global Average Pooling Layer to OperationResolver.");
            setUpSuccessful = false;
        }

        if (statusFC != kTfLiteOk)
        {
            dsp_print_error("Failed while adding FullyConnected Layer to OperationResolver.");
            setUpSuccessful = false;
        }

        return setUpSuccessful;

```
I would be really greatful if someone were to help fix this issue.

oniigiirii, 2023-08-29:
Hello
I am currently facing the same issue with a model I am trying to run on a dsp platform.
Error:
```
Didn't find op for builtin opcode 'MEAN'
Failed to get registration from op code MEAN
```
This is my resolver operation registration code:
```
        auto statusConv2d = resolver.AddConv2D();
        auto statusDepthwiseConv2d = resolver.AddDepthwiseConv2D();
        auto statusRelu = resolver.AddRelu();
        auto statusFC = resolver.AddFullyConnected();
        auto statusMean = resolver.AddMean();

        /// Verify Resolver Operation addition
        bool setUpSuccessful = true;

        if (statusConv2d != kTfLiteOk)
        {
            print("Failed while adding Conv2d Layer to OperationResolver.");
            setUpSuccessful = false;
        }

        if (statusDepthwiseConv2d != kTfLiteOk)
        {
            print("Failed while adding DepthwiseConv2d Layer to OperationResolver.");
            setUpSuccessful = false;
        }

        if (statusRelu != kTfLiteOk)
        {
            print("Failed while adding ReLU Activation Function to OperationResolver.");
            setUpSuccessful = false;
        }

        if (statusFC != kTfLiteOk)
        {
            print("Failed while adding FullyConnected Layer to OperationResolver.");
            setUpSuccessful = false;
        }

        if (statusMean != kTfLiteOk)
        {
            print("Failed while adding mean Layer to OperationResolver.");
            setUpSuccessful = false;
        }

        return setUpSuccessful;
```

Would be really grateful if someone were to help fix this issue.

lioff, 2023-09-06:
I was able to fix it! My approach to adding operations to the resolver no longer works.
The following code correctly registers the operations:

```
namespace {
using HelloWorldOpResolver = tflite::MicroMutableOpResolver<7>;

TfLiteStatus RegisterOps(HelloWorldOpResolver& op_resolver) {
  TF_LITE_ENSURE_STATUS(op_resolver.AddFullyConnected());
  TF_LITE_ENSURE_STATUS(op_resolver.AddQuantize());
  TF_LITE_ENSURE_STATUS(op_resolver.AddReshape());
  TF_LITE_ENSURE_STATUS(op_resolver.AddConv2D());
  TF_LITE_ENSURE_STATUS(op_resolver.AddMaxPool2D());
  TF_LITE_ENSURE_STATUS(op_resolver.AddSoftmax());
  TF_LITE_ENSURE_STATUS(op_resolver.AddDequantize());

  return kTfLiteOk;
}
}  // namespace
```

And:

```
HelloWorldOpResolver op_resolver;
RegisterOps(op_resolver);
```
(RegisterOps should also be checked for errors; see [hello_world_test.cc](https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/examples/hello_world/hello_world_test.cc)).

closed 2023-09-06 by lioff