tensorflow/tflite-micro #2207: General arduino port of TFLite-micro?

sussman, 2023-09-04:
Hi all!  I'm working my way through the "TinyML" O'Reilly book (by Pete and Daniel).  I already own a couple of modern arduinos (a Adafruit Bluefruit Sense feather, and Adafruit ESP32-S2 Feather), and I'm surprised that TFLite only works on a single specific Arduino (the Nano 33 BLE Sense).  

The front page of the project cheerfully suggests that TFLiteshould "work on any arm-m4-cortex" device (like my Bluefruit Sense), but after a couple of hours I gave up trying to get the "hello world" sine-wave example to compile in the IDE for my feather.  I commented out the #ifdef test that looks for the specific 33 BLE model, but like so many others, got lost in a maze of min() and max() being defined too many times.

So I'm asking:  is anyone working on a more generalized m4-cortex port?  Perhaps one that *doesn't* support all the custom sensor hardware for the 33 BLE Sense device, but is simply guaranteed to be able to do nothing but run the TF Interpreter?

hpssjellis, 2023-09-16:
@sussman I also would like to know how to port tFlite-micro to any of the hundreds of microcontrollers that could probably do TinyML. Years ago I used to have a very simple way to do it, but that has been deprecated. The breaking point used to be logging debug info. Has that changed?

All of the single page Arduino examples on this [repository folder](https://github.com/hpssjellis/my-examples-for-the-arduino-portentaH7/tree/master/m09-Tensoflow) used to run on multiple devices. (I was mainly using the then new  Arduino PortentaH7).   The main part of my code was a bunch of if statements that chose the correct output method for the device. Can someone suggest a way to include this method in the new version of tFlite-micro?  My very bad solution was to download an old version of tflite-micro that I had changed slightly. I would prefer to run a new version of tflite-micro. Any suggestions for a fix.

@rkuester  @mansnils 

Here is what used to work

```

// Start Debug.cc information    ----------------------------------------------------------

 #if  defined (CORE_CM7)   || defined (CORE_CM4)  || defined (TEENSYDUINO) ||  defined (ARDUINO_NANO33BLE) ||  defined (YOUR_BOARD1) // CORE_CM7 is for the PortentaH7 outer core
 
    // do obsoletely nothing the default works

 // Note:  NANO_33_BLE deprecated as MBED of 1.30, new name: ARDUINO_NANO33BLE  
 // Note:  CORE_CM4 MBED of 1.30, moved above  
 #elif   defined (NANO_33_BLE)   ||  defined (YOUR_BOARD2)  // CORE_CM4 is for the PortentaH7 inner core
 
     #define DEBUG_SERIAL_OBJECT (Serial) 

     extern "C" void DebugLog(const char* s) {
        static bool is_initialized = false;
        if (!is_initialized) {
           DEBUG_SERIAL_OBJECT.begin(9600);
           is_initialized = true;
        }
        DEBUG_SERIAL_OBJECT.print(s);
     }

 #elif defined (__SAM3X8E__)  ||  defined (YOUR_BOARD3) // Arduino UNO style boards

     #define DEBUG_SERIAL_OBJECT (SerialUSB) 

     extern "C" void DebugLog(const char* s) {
        static bool is_initialized = false;
        if (!is_initialized) {
           DEBUG_SERIAL_OBJECT.begin(9600);
           is_initialized = true;
        }
        DEBUG_SERIAL_OBJECT.print(s);
     }
 #elif defined (SEEED_XIAO_M0)  ||  defined (YOUR_BOARD4) // The new $5 USD Seeeduino XIAO board

     #define CFG_TUSB_DEBUG
     
 #else
   // don't do any debugging until you figure out your board  
   
   extern "C" void DebugLog(const char* s) {
     // Do not log debug info
   } 
 #endif

/*
 * nano_33_iot.build.board=SAMD_NANO_33_IOT  // have not got it working yet
 * 
*/

// End Debug.cc information    ----------------------------------------------------------

```

This 2021 comment by @advaitjain  seems to address this issue, https://github.com/tensorflow/tflite-micro/issues/407#issuecomment-898660589   and for Arduinos the solution seems to be this repo

https://github.com/tensorflow/tflite-micro-arduino-examples

but that repo has not been supported for a while.  Back to the original questions, is there an easy way to port TFlite-micro to other microcontrollers, like my list of if statements above?

P.S. Why did I leave tFlite-micro 3 years ago, because [edgeimpulse.com](https://edgeimpulse.com/) made it all so easy and pure tFlite-micro was not needed. I would now like to get back to tFlite-micro because my needs and microcontrollers have changed.

@petewarden

hpssjellis, 2023-09-17:
So this is very interesting on the [tflite-micro-arduino-examples](https://github.com/tensorflow/tflite-micro-arduino-examples)" for the Nano33BleSense  using the Tensorflow library 

Note: Which loads as Arduino_TensorFlowLite  as if it is a library written by Arduino. 

https://github.com/tensorflow/tflite-micro-arduino-examples/blob/76052fc778fb3638667d69886503733653550bfc/src/peripherals/peripherals.h#L34

In the file at 

[https://github.com/tensorflow/tflite-micro-arduino-examples/src/peripherals/peripherals.h](https://github.com/tensorflow/tflite-micro-arduino-examples/blob/main/src/peripherals/peripherals.h)

```
#if defined(ARDUINO_ARDUINO_NANO33BLE)
#include <cstdint>

#include "button.h"
#include "led.h"
#include "ws_wm8960_audio_hat_nrf52840.h"

#define AUDIO_DEVICE_WS_WM8960_AUDIO_HAT \
  &peripherals::WS_WM8960_AudioHat_NRF52840::Instance()

namespace peripherals {

constexpr PinName kI2S_BIT_CLK = P0_27;   // D9
constexpr PinName kI2S_LR_CLK = P1_2;     // D10
constexpr PinName kI2S_DATA_IN = P1_12;   // D3
constexpr PinName kI2S_DATA_OUT = P1_11;  // D2
constexpr uint32_t kI2S_IRQ_PRIORITY = 7;

constexpr uint32_t kI2C_CLOCK = 100000;

constexpr pin_size_t kBUTTON_GPIO = D8;

constexpr pin_size_t kLED_DEFAULT_GPIO = D13;

}  // namespace peripherals

#else  // ARDUINO_ARDUINO_NANO33BLE
#error "unsupported board"

#endif  // ARDUINO_ARDUINO_NANO33BLE
```

hpssjellis, 2023-10-12:
I think this is still an important issue.

sussman, 2023-10-16:
Six weeks later:  Is this project maintained anymore?   Anyone looking at issues?  Or is there perhaps a mailing list?

rascani, 2023-10-16:
> Hi all! I'm working my way through the "TinyML" O'Reilly book (by Pete and Daniel). I already own a couple of modern arduinos (a Adafruit Bluefruit Sense feather, and Adafruit ESP32-S2 Feather), and [...]

I see no reason why this shouldn't work, even if it only uses reference kernels. However, as I mentioned on #2229, we're not actively maintaining the arduino examples repository. Could you give more specifics about how you're building the project and what issue you're seeing?

> @sussman I also would like to know how to port tFlite-micro to any of the hundreds of microcontrollers that could probably do TinyML. Years ago I used to have a very simple way to do it, but that has been deprecated. The breaking point used to be logging debug info. Has that changed?

Yes, this changed not too long ago in PR #2073, which pushed the responsibility of print formatting onto the platform implementation. The signature for DebugLog now includes format args:
```
external "C" void DebugLog(const char* format, va_list args);
```
If your platform doesn't have any print formatting, you can also use the eyalroz_printf library like we do on the [bluepill platform](https://github.com/tensorflow/tflite-micro/pull/2073/files#diff-b5fa86a50a1efa2885c672fecc2e9af372e354c511c89fbfcf143d346b3f4b10).

AIWintermuteAI, 2024-09-14:
So, after spending a few hours trying (unsuccessfully) build a project using 
https://github.com/tensorflow/tflite-micro-arduino-examples (the official library)
I found this fork
https://github.com/spaziochirale/Chirale_TensorFlowLite
which seems to be maintained. Compiled and run without issues on ESP-EYE (ESP32).
Just leaving it here for someone to find if struggling with official abandoned library.

sussman, 2024-09-16:
Hey thanks, I'll give it a try when I have time!