tensorflow/tflite-micro #2229: Porting to other boards.

hpssjellis, 2023-09-17:
Hi:  @advaitjain @mansnils @rascani @rkuester 

I would like to port this code to a few other boards like I did 3 years ago [here](https://github.com/hpssjellis/my-examples-for-the-arduino-portentaH7/tree/master/m09-Tensoflow). Most of the boards are Arduino: the PortentaH7, Nano33BleSense REV2, Nicla boards, but I am also interested in the Seeedstudio XIAO boards and also testing Coral-Micro on the Arduino IDE (which presently has serial read issues) . I do see the Tensorflow github  https://github.com/tensorflow/tflite-micro-arduino-examples which I have replied to in this issue [General arduino port of TFLite-micro?](https://github.com/tensorflow/tflite-micro/issues/2207#issuecomment-1722252714)

Just wondering who are the leads on the tflite-micro-arduino-examples? Is it still active? If so are you interested in any PR's? 

In the past I have had a good relationship with Google when they helped me with my massive list of [Rocksetta TensorflowJS](https://hpssjellis.github.io/beginner-tensorflowjs-examples-in-javascript/) single webpage ML examples.

closed 2023-10-13 by ddavis-2015

hpssjellis, 2023-10-13:
@ddavis-2015 thanks for replying and that does help a bit (I am most interested in the PortentaH7, and NiclaVision) , but if the [Arduino tFlite micro repo](https://github.com/tensorflow/tflite-micro-arduino-examples) is not being supported by Google I might as well use my awesome 3 year old tflite micro version [here](https://github.com/hpssjellis/RocksettaTinyML).

Is there a way we can work with Google to keep supporting the Arduino Library? I have also been using the CoralMicro dev board and it has an Arduino port which is not working very well yet, see issue 72 [here](https://github.com/google-coral/coralmicro/issues/72) . Can these two Arduino issues be made into one issue? Could the CoralMicro repo be fixed and adapted to work with other boards? I could help test the CoralMicro on the Arduino IDE with my massive  Arduino examples library the portenta-pro-community-solutions [here](https://github.com/hpssjellis/portenta-pro-community-solutions)

I would be willing to work on both issues to try to improve Arduino IDE support by Google.

rascani, 2023-10-16:
Hi @hpssjellis. The Arduino library was originally frozen due to incompatibilities between the Arduino GCC compiler and the CMSIS NN library. If you're interested in pursuing this, I'm happy to help advise and review/approve PRs, but we don't currently have the bandwidth to actively maintain it.

I can't speak to the CoralMicro project, though my understanding is they're also on an old port of TFLM.

hpssjellis, 2023-10-16:
> Hi @hpssjellis. The Arduino library was originally frozen due to incompatibilities between the Arduino GCC compiler and the CMSIS NN library. If you're interested in pursuing this, I'm happy to help [...]

Thanks for the reply @rascani, really interesting that CoralMicro might be using the old port. As far as the CMSIS NN library I feel that [edgeimpulse.com](https://edgeimpulse.com/) has solved that problem so I am curious what might need to be done.

@rascani are you familiar with TensorflowJS, and Arduino microcontrollers? I am in the process of merging the two concepts and can do it with a 3 year old TFLM Arduino port but it would be much better to use a new TFLM Arduino Port.

I teach High School Robotics and do a lot of projects:  https://github.com/hpssjellis