tensorflow/tflite-micro #2231: How to make tflm

AngLi666, 2023-09-19:


closed 2023-10-19 by github-actions[bot]

Dimfred, 2024-01-30:
Hello,
I am running into the same issue here.
Is there a workaround for this at least?

Code to reproduce:

```python
from pathlib import Path
import keras
import numpy as np
import tensorflow as tf
from keras.layers import LSTM, Dense, RepeatVector, TimeDistributed
from keras.models import Sequential

class Model(keras.Model):
    def __init__(self):
        super().__init__()

        self.model = Sequential()
        self.model.add(
            LSTM(
                32,
                activation="relu",
                input_shape=(n_timesteps, n_features),
                return_sequences=True,
            )
        )
        self.model.add(LSTM(16, activation="relu", return_sequences=False))
        self.model.add(RepeatVector(n_timesteps))
        self.model.add(LSTM(16, activation="relu", return_sequences=True))
        self.model.add(LSTM(32, activation="relu", return_sequences=True))
        self.model.add(TimeDistributed(Dense(n_features)))
        self.model.compile(optimizer="adam", loss="mse")
        self.model.build()
        self.model.summary()

    def call(self, inputs):
        return self.model(inputs)

model = Model()
model.predict(tf.random.normal((1, n_timesteps, n_features)))

concrete_func = run_model.get_concrete_function(
    tf.TensorSpec([1, n_timesteps, n_features], "float")
)
model.save("keras.tf", save_format="tf", signatures=concrete_func)

converter = tf.lite.TFLiteConverter.from_saved_model("keras.tf")
converter.target_spec.supported_types = [tf.float16]
converter.inference_input_type = tf.float32
converter.inference_output_type = tf.float32

tflite_model = converter.convert()
tflite_model_path = Path("keras.tflite")
tflite_model_path.write_bytes(bytes(tflite_model))
```

Would be happy about any suggestions.

AngLi666, 2024-01-31:
It is dead way to work with TFLM.
You can try TVM if appliable to you

AIWintermuteAI, 2024-10-01:
@Dimfred @AngLi666 Edge Impulse fork of tflite micro has Tile OP ported.
Disclaimer: I work for Edge Impulse and did the porting :)

fiti-tsunche, 2025-05-14:
Does Edge Impulse open source your fork of tflite micro?

AIWintermuteAI, 2025-05-19:
It's open source, you can find the code inside edge-impulse-sdk folder in C++ deployment - any project will do, the sdk is the same.
But for the question you asked - I think there are cases when TILE works with fixed output size and only those can be supported by tflite micro, as tflite micro does not support dynamic tensor sizes (or at least it didn't last time I checked - [here](https://github.com/tensorflow/tflite-micro/blob/9d35a74b03fa313026982abb4d2c4ffe29de88bc/tensorflow/lite/micro/kernels/batch_matmul_common.cc#L65) and some other comments in the code base)
What I did there was 
```
  if (IsDynamicTensor(output)) {
    TF_LITE_KERNEL_LOG(context, "Dynamic output tensor is not supported by tflite micro.");
    micro_context->DeallocateTempTfLiteTensor(input);
    micro_context->DeallocateTempTfLiteTensor(output);
    micro_context->DeallocateTempTfLiteTensor(multipliers);
    return kTfLiteError;
  }
```

fiti-tsunche, 2025-05-19:
@AIWintermuteAI Thanks for your reply!

It seems I have to deeply understand tflite micro's memory management policy to make it support dynamic tensors. It would be difficult and dangerous.