tensorflow/tflite-micro #2232: tflite 8-bit quantization question

surez-ok, 2023-09-19:
Hello，friends：

I am trying to convert h5 model to tflite-int8 model，the code is as follows:
~~~
model_filename = "alexnet_tflite_kws_model.h5"
tflite_filename = "alexnet_tflite_int8.tflite"
model = models.load_model('./weight/' + model_filename)

converter = tf.lite.TFLiteConverter.from_keras_model(model)

converter.optimizations = [tf.lite.Optimize.DEFAULT]
converter.representative_dataset = representative_dataset_gen

converter.target_spec.supported_ops = [tf.lite.OpsSet.TFLITE_BUILTINS_INT8, tf.lite.OpsSet.SELECT_TF_OPS]

#converter.inference_input_type  = tf.int8

#converter.inference_output_type = tf.int8

quantiz_model = converter.convert()

open('./weight/' + tflite_filename, "wb").write(quantiz_model)
~~~

At end, I get the int8 tflite model successfully, but it will get wrong answer when use the model to inference (tflite f32 model is OK but tflite int8 has large error)

I wander that: 
1. Is there only one way to  convert model to tflite-int8 model as shown above?
2. Is there some ways to debug the problem when meeting large error like me,

avptin, 2023-09-25:
Hi!

1. The only way to convert is to use convert() method and it works in most cases. For the parameters, check https://www.tensorflow.org/lite/performance/post_training_quantization.

2. I would check the representative calibration data: are you getting large error when running inference on the calibration data?

Burton2000, 2023-09-26:
Quantization aware training can help sometimes if the error from post-training quantization is too much. See: https://www.tensorflow.org/model_optimization/guide/quantization/training

surez-ok, 2023-09-28:
thanks，guys.  

I have tried many times and found that it will product large error if the representative data has uneven distribution. I  will train other model to handle this case.  

I will close this issue, thanks.

closed 2023-09-28 by surez-ok